ethz-asl/maplab #86: How should I use vi-map to localize without IMU messges?

wendaoyuchen, 2018-08-09:
Hi , I am trying to localize with a single image in a stored vi-map.
I am not very familiar with maplab's data.
I found there are some flow(datasource_flow_,rovio_flow_,localizer_flow_ and synchronizer_flow_), I guess what I need is localizer_flow_ , but I am confused how IMU and image data flow into it .
And actually I want the input is just images.
Can you give me some suggestions?

dymczykm, 2018-08-09:
Hi @wendaoyuchen,
we're afraid you can't localize a single image within a stored vi-map. You would need to detect the features first and create an NFrame and then pass it to the localizer engine, here:
https://github.com/ethz-asl/maplab/blob/master/algorithms/loopclosure/loop-closure-handler/src/loop-detector-node.cc#L490

We would suggest to just use a second recording (e.g. a rosbag) with an IMU to perform localization as suggested in the wiki.

wendaoyuchen, 2018-08-10:
Thanks for your suggestions @dymczykm ! What I want is to verify some relocalization algorithms in maplab framework.So the first thing I want to do is to extract the relocalization module of maplab and make it work. I guess the system should be like below:
_____________                    ________                   ________
|  image frame   |   --------->   |   vi-map |  --------->   |    R, t      |
----------------------                    -------------                   -------------
As far as I know, relocalization module is based on the image information without IMU. However, I am not familiar with the relocalization algorithm in maplab, I am confused about the input of the localizer, I guess the input is `aslam::VisualNFrame::Ptr nframe` which contains a list of frames. Why does relocalization module need several frames not one frame?
Can you give a link of papers to introduce the algorithms in relocalization module?

dymczykm, 2018-08-16:
@wendaoyuchen you're right, the localizer requires just a frame, no IMU. The ``NFrame`` type is basically a collection of one or multiple ``Frame`` objects so one is enough (we wanted to generalize it to multi-camera systems). So basically you would need to construct such a ``Frame`` object, embed it within ``NFrame`` and feed it to the localizer.

For the papers, you can take a look at the following list:
https://github.com/ethz-asl/maplab/wiki/Related-Research
Especially "Get Out of My Lab" paper might be very relevant.

wendaoyuchen, 2018-08-22:
@dymczykm Thanks !
I did some changes to make rovioli can accept ROS bag contained image messages only, and it can output the result of localization. But I am not sure what I did is right.
in rovioli-node,I change 
` datasource_flow_.reset(new DataSourceFlow(*camera_system, *maplab_imu_sensor))`  to 
`datasource_flow_.reset(new DataSourceFlow(*camera_system));` ;
`tracker_flow_.reset(
        new FeatureTrackingFlow(camera_system, *maplab_imu_sensor));`   to
`tracker_flow_.reset(
                new FeatureTrackingFlow(camera_system));`  ;
delete the rovio_flow and change the synchronizer_flow_  
......
I want to know if it is neccessary to change so lot ? just close the imu message in these flows ,is that OK?

dymczykm, 2018-08-27:
@wendaoyuchen I would rather create a simpler flow yourself (taking the current one as a reference only) that just consumes the images from the rosbag, creates NFrames, does feature tracking and feeds the NFrames with keypoints to the Localizer. 

Please bear in mind that the tracker requires the IMU data in ``SynchronizedNFrameImu`` which you should probably fill with some dummy timestamps and zeros as IMU readings.