ethz-asl/maplab #87: the time of opencv_catkin build reaches a day ? so long long time?

fangxu622, 2018-08-10:


schneith, 2018-08-10:
The opencv_catkin package downloads the source code before building. Sometimes this downloads fails and let's the build hang. In that case, it is best to clean your workspace with 'catkin clean' and restart the build.

fangxu622, 2018-08-10:
@schneith   I downloaded opencv3 sourcecode before building.This situation still occurs

ZacharyTaylor, 2018-08-10:
How much ram does your computer have? If it is running out (sometimes an issue with < 4gb) this could cause the issue. You could also try building with -j 1 to ensure only one package builds at once.

fangxu622, 2018-08-10:
RAM  = 8gb

fangxu622, 2018-08-10:
@ZacharyTaylor  ram 8gb

fangxu622, 2018-08-15:
@dymczykm  yes, catkin clean , then input "yes" or  "no", We've tried.it always stucked in opencv3_catkin build phase， Sometimes it lasts 4 days in opencv3_catkin build phase .
[image]

fangxu622, 2018-08-15:
[image]

mfehr, 2018-08-15:
Hi @fangxu622 

Please clean and rebuild the workspace again, but this time use the following flags:
```bash
catkin build opencv3_catkin -v --interleave-output
```
This will help you to figure out exactly where the build gets stuck. As @schneith  already pointed out, opencv3_catkin can cause trouble if you have a weak internet connection, because it takes a while to download or there might be a timeout. But if the build takes more than an hour on a modern machine it is most likely stuck. So no need to wait for 4 days :)

smauq, 2018-08-15:
Another possibility from the [FAQ](https://github.com/ethz-asl/maplab/wiki/FAQ) on "Why is my opencv3_catkin build freezing?". As ridiculous at it sounds, apparently this can be an issue:

> A known cause for this is an installed MATLAB using a network licence. The build can stall on getting the licence authenticated; so make sure that the licence server can be accessed, connect to the VPN if required."

fangxu622, 2018-08-15:
@mfehr  thank you ,now ,this problem occurs, no "opencv_src-configure" files.   the opencv source code is downloaded before buiding,  Is there supposed to be  "opencv_src-configure" file？
[image]

[image]

fangxu622, 2018-08-15:
[image]

fangxu622, 2018-08-15:
@smauq  it 's network problems ,but something else wrong after fixing network problems

fangxu622, 2018-08-15:
ok, fixed it,thanks all @schneith @ZacharyTaylor @dymczykm @mfehr ,just retry

dbHub20092758, 2022-03-30:
the same problem , how to fix it. i download opencv package before build @fangxu622