ethz-asl/maplab #89: Nobody notices the memory leak in vi-map-optimizer.cc ?

NewThinker-Jiwey, 2018-08-15:
The variable [optimization_problem](https://github.com/ethz-asl/maplab/blob/2db80eb0c1d56e4077edffcf43bbce02deb8a046/algorithms/map-optimization/src/vi-map-optimizer.cc#L57) is allocated in ```VIMapOptimizer::optimizeVisualInertial()```, but not been freed anywhere.
Is it necessary to create a pull request for this ?

dymczykm, 2018-08-15:
@NewThinker-Jiwey You're right, #90 should fix that

closed 2018-08-15 by dymczykm

NewThinker-Jiwey, 2018-08-22:
@dymczykm  Nice work !

But I might not have reported that bug completely. A similar mistake is still in the ```vi-map-relaxation``` part, see [here](https://github.com/ethz-asl/maplab/blob/2db80eb0c1d56e4077edffcf43bbce02deb8a046/algorithms/map-optimization/src/vi-map-relaxation.cc#L116)

PR #90 (2018-08-15): fixed a leak, optimization problem was not freed
Fixes #89
The object``map_optimization::OptimizationProblem`` was never destructed which resulted in a leak every time the optimization was run.