ethz-asl/maplab #93: Error while installing maplab_common?

Keudeugi, 2018-09-09:
Hi, first of thanks for providing this awesome program!

I'm currently getting problems while installing maplab, and it seems related with installation of module: maplab_common.

Specs: Ubuntu 16.04, ROS Kinetic, Catkin workspace (User spec: I started working with Ubuntu from three days ago; Newbie)

In the installation step, though I already installed ROS kinetic & catkin, I followed all of the commands provided in the page.
Everything seemed as work except one command; **catkin build maplab**
A bunch of packages failed to be installed, following issue shows similar result about the error: [https://github.com/ethz-asl/maplab/issues/92](url)

As I examined build log, there was an error seemed related with Cmake: CMake Error at /home/piljoong/maplab_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:38 (find_package)
After this error I met list of failed packages:
[image]

Did I miss a step while installing the project? Can I provide an additional informations for figuring this out? Thanks in advance.

PS. Here's my catkin config:
[image]

mfehr, 2018-09-09:
Hi @Keudeugi ,
Can you check if aslam_cv2 (and the package aslam_cv_common) is actually there? It's a submodule of maplab and if you forgot to use the `--recursive` flag when cloning the repo or didn't call `git submodule init` / `git submodule update` at some point it might not be there.

Keudeugi, 2018-09-10:
Oh my, what a shame. Maybe I cloned submodules only for maplab_dependencies. Thank you!

closed 2018-09-10 by Keudeugi

karankatiyar92, 2021-02-22:
> Hi @Keudeugi ,
> Can you check if aslam_cv2 (and the package aslam_cv_common) is actually there? It's a submodule of maplab and if you forgot to use the `--recursive` flag when cloning the repo or didn't call `git submodule init` / `git submodule update` at some point it might not be there.

i have those both still i get the same issue ?, any suggestions how can i fix it