ethz-asl/maplab #100: MapLab Unstable with Zed Mini

rhklite, 2018-10-04:
Hi all, please excuse me if my questions are naive, I just started working with ROS. I'll try my best to describe my issue, and have bolded my questions in text.

## Issue Seen
I'm currently trying to get the Zed mini (camera with built in imu) working with Maplab, but currently encountered 2 types of problems:
1. Maplab diverges at the start.
2. Does not diverge at the start, but outputs a lot of the following messages. Usually the OutlierDetection message shows up the most, then many feature-tracking error messages would popup.
    - W1003 22:16:34.001222 22614 OutlierDetection.hpp:47] Detected outlier - mahalanobis (d / threshold):9.82983 > 9.21 - count : 1.
    - W1003 22:18:17.438231 25341 feature-tracking.cc:35] No bias from the estimator available. Assuming zero bias.

## Change made to Maplab Code
I had to made a change to the maplab code. Though I'm not sure what is the impact of this change.

I commented out this section of code due to Zed Mini uses the same timestamp for all the imu messages associated with a image frame, thus causing maplab to crash. https://github.com/stereolabs/zed-ros-wrapper/issues/243#issuecomment-412649684 

This practise was done by another person who had a similar issue with maplab and Tango https://github.com/ethz-asl/maplab/issues/48#issuecomment-370432552
```
// ~/catkin_ws/src/MapLab/maplab/common/vio-common/include/vio-common/internal/imu-measurements-buffer-inl.h
// commented code:
// vio::ImuMeasurement last_value;
// if (buffer_.getNewestValue(&last_value)) {
// CHECK_GT(timestamp_nanoseconds, last_value.timestamp)
// << "Timestamps not strictly increasing.";
// }
```
## Suspected Cause of Issue:
**Bad Calibration: Bad sensor or bad calibration process?**
- the reprojection error of my kalibr calibration result consistently fall in the range of 1-2px for 1280x720@60fps or around 0.5px for 672x376@100fps. Whereas github issue post and kalibr sample calibration suggests that 0.1-0.2px is considered good.
- My imu/gyro error is significantly higher than the sample calibration result.
- I tried different fps, motion profile, lighting, calibration target, but nothing has made any significant change to the reprojection error or accel/gyro error.
- **I'm not sure if this means my calibration process is faulty or the sensor itself is not good.**

**Bad Timestamp**
- Because I forced Maplab to bypass the imu stamp strictly increasing check. I'm not sure how this would've affected ROVIOLI.
- I've tried running Maplab When ZedMini is using both the sync'd timestamp and ros time stamp. However no change in Maplab's performance was seen.
- **How would this particular time stamping method that Zed Mini is using affects Maplab?**
- **Are there are any mechanisms in rovio/maplab that deals with this?**
- Kalibr_calibration_imu_camera is able to compute a time-offset between the camera frame and the imu frame. **Would pass this time-offset help with the problems i'm seeing?** If yes, **how can this be done?**

Davide-Rigamonti, 2018-10-26:
Hi rhklite, I've noticed out that you seem working on Visual Based Slam. I have some questions about what are you facing dealing with these freameworks and cameras. I'm looking towards Visual Slam as well. Can I contact you in some way? Thank you!!

rhklite, 2018-10-26:
@Davide-Rigamonti sure thing. I'll shoot you an email

salahkhan94, 2018-11-10:
I tested Maplab with Zed Camera and Xsens mti-1 , I've faced exactly all the issues that you've faced. Though I've rarely encountered rovioli diverging at the start but it definitely diverges 3-4 min after and starts throwing up "No bias from the estimator available. Assuming zero bias."

Thinking that it could be because of calibration issues , I calibrated 3 times and arrived at the same calibration values each time, Kalibr also doesn't detect any time-offset. 

Suspects:
1)The Orientation values look pretty good but there are some anomalies in Translation between IMU and Camera and I don't think it is that big of an issue.  

2)IMU specs are lower than ADIS16488 and the actual bias and noise could be higher than what Kalibr estimates but again if rovio can work with MPU-9250  then it should also work well with Mti-1 of xsens ,or with ZED mini.

3) Smooth and steady motion https://github.com/ethz-asl/maplab/issues/54#issuecomment-373117878
this looks to be a probable cause, perhaps integrating wheel odometry could alleviate bias issues.

4) Zed camera by default dynamically changes the Contrast and exposure of the camera depending on the lighting condition.

@rhklite  @Davide-Rigamonti  I'm trying to solve the same problem perhaps we could collaborate.

salahkhan94, 2018-11-11:
Yes I'm using http://wiki.ros.org/xsens_driver and Kalibr gives exactly 0ms as the time-offset, I have also faced a more aggravated divergence with msckf_vio with the same sensor-setup right at the first instance of motion.  I guess then you're right as the IMU does do some heavy filtering onboard. 

Is there any other way to estimate the time-offset? Or I'd have to somehow publish the un-filtered IMU readings and then calibrate to estimate the time-offset.

Thanks.

rhklite, 2018-11-11:
@salahkhan94 I ended up using other packages since I couldn't get maplab to work. 
I suspected MAPLAB requires imu and camera frame to be hardware synchronized, or there would be time-stamp issues. https://github.com/ethz-asl/rovio/wiki/Divergence

rhklite, 2018-11-29:
@mfehr No problem, thanks for letting me know!

closed 2018-11-29 by rhklite