ethz-asl/maplab #99: ROVILIO has scale and robustness problems

maartenvlaswinkel, 2018-10-04:
Hi,

I am using a drone setup with an IMU that is sampling at 500Hz and a rolling shutter camera sampling at 30Hz. I have been trying to come up with the proper values for the noise parameters. At first I tried to find these parameters using the Allan variance, however, these values result in a poor behavior. 

Now, I am trying to tune the variables manually and I noticed several things:

- The bias, or random-walk, parameter influences the scaling. When the bias is set 'wrong', the range of motion determined by ROVIOLI is enlarged compared to the actual movement. At best I am getting a range of motion of around one meter when it should be around 0.5 meters.
- ROVIOLI tends to become unstable when the setup stops moving. This especially becomes a problem when the above mentioned scaling is getting closer to what it should be.

In these test I try to reduce motion blur and shocks as much as possible.

Is there anybody who has some tuning guidelines for ROVIOLI or who knows what cause my observations?

mfehr, 2018-10-16:
Hi @maartenvlaswinkel

There are many reasons for scale and robustness issues:

 * Bad/no time synchronization of the IMU to the cameras
 * Calibration of the camera and IMU
 * Camera settings
 * Rolling shutter (there is no compensation in maplab for this)

Assuming the accuracy of the calibration and sensor synchronisation is sufficient, we did observe similar issues as you describe it when tuning the IMU parameters for new sensor, especially with poorly calibrated IMUs (commonly with less expensive sensors).
Your observation is correct with respect to the influence of the IMU parameter on the scaling. One reason is that if you change the parameters you change the 'weighting' between camera and IMU, which can affect scale a lot, since the IMU is the only source for scale. In extreme cases where the IMU is not used, you don't get any scale information.

Unfortunately we don't have any guidelines that always work, there is always some manual tuning involved, but the Allen invariance is already a good starting point. You can also have a look at https://github.com/gaowenliang/imu_utils to analyse your IMU performance.

Could you share what kind of sensor setup you are using?

maartenvlaswinkel, 2018-10-18:
Hi @mfehr,

Thank you for you response.

I discovered that there is a major problem with the synchronization of the IMU and the camera and I am now trying to fix this issue.

I am using a Crazyflie 2.0 with a MPU-9250 IMU sensor and a cheap FX797T rolling shutter camera. I know that this is a far from ideal setup, but it is part of the challenge we have set for ourselfs. 

I was also wondering if it is possible to use other sensors to retrieve a proper scaling, for example an optical flow sensor. Do you know where this should be implemented into ROVIO(LI)?

maartenvlaswinkel, 2018-11-12:
Hi,

I was able to solve the problem I described in the first post. The major issue was the synchronization between the IMU and camera. After solving this, I got a decent result using the noise parameters found using the Allan variance and with some tuning I was able to improve the results even more.

Another thing that seemed important, especially at the start of the estimation, was the initialization depth of the features. In my scene, most features where closer to the camera then the default distance. By changing this value I was able to get a proper estimation from the start. Increasing the initial variance of the feature, however, did not always result in a better result and could even result in a estimation failure.

closed 2018-11-12 by maartenvlaswinkel