ethz-asl/maplab #103: The rule to change "hardware_id" or "label" in yaml files

huabit, 2018-10-15:
Hi all, I'm currently trying to get the Zed mini (Stereo Camera with built in imu) working with Maplab, but encountered some problems, I'll try my best to describe my issues, and have bolded my questions in text.

**How to change hardware_id or label in yaml files**
---
As I know that Maplab needs 3 yaml files [Sensor-Calibration-Format](https://github.com/ethz-asl/maplab/wiki/Sensor-Calibration-Format) which are discribing cameras, imu-maplab and imu-rovio. I'm interested with yaml files for cameras and imu-maplab. There are two parameters making connections between ROStopics published by the Zed mini and the subscriber of Maplab. One is the "label" in cameras yaml file, the other is the "hardware_id" in imu-maplab yaml file. Which are listed blow:
 - id: 8af1bac2a7c5fad79f09e49e3b96d87c
cameras:
   -camera:
      **label: /zed/left/image_raw_color** 
      id: ac4704b6b1b4cf41b0924ecb7bd863ec
      line-delay-nanoseconds: 0
      image_height: 720
      image_width: 1280
      type: pinhole

 - id: 01141d0b32bd6ba2c5f254ae8da087d3
**hardware_id: /zed/imu/data**
sensor_type: IMU
sigmas:

Where the "label" and the "hardware_id" meeting the exact ROStopic names published by ZEDmini. But it seems doesn't work. 

So, my question is the rules of Maplab to name the "label:" and the "hardware_id:", and how can I follow the rules to change in order to establish connections between ROStopics published by the Zed mini and the subscriber of Maplab.
Is it necessary to modify the source code of maplab? If yes, how can this be done?

---
Expecting any reply. Thanks in advance!

mfehr, 2018-11-29:
Hi @huabit 
Not sure I can follow exactly what the issue is. So if you want to subscribe to different ros topics all you need to do is:
 * change the `label` in the camera yaml to your rostopic, however we always append `*/image_raw` to the rostopic (as described [here](https://github.com/ethz-asl/maplab/wiki/Running-ROVIOLI-in-VIO-mode#building-a-map-from-a-rosbag)), so if your rostopic is: `/mynode/camera/best_image_ever` you can set the rostopic in two different ways:
   * set `label` to `mynode/camera/best_image_ever` and `-vio_camera_topic_suffix=""`
   * set `label` to `mynode/camera` and `-vio_camera_topic_suffix="best_image_ever"`

 * change the `hardware_id` to your imu topic (you should be able to discard the initial `/`)

I know this is not very straight forward, the fact that we have two different parameter names that are not explaining even in the slightest, that these are the ros topic isn't all that great. These parameter files just evolved from legacy systems and nobody has bothered to change them because it would break some interfaces.

So short answer: Yes, you can change the ros topics in the yamls only. No, you don't need to change maplab code.

PS: seeing that you are probably using the zed mini, I just want to give you a heads-up, if you have the same sensor we have, it does not use proper hardware timestamping, the ROS node just assigns timestamps based on arrival time (ROS time or image arrival time in USB buffer or sth like this). We did manage to get some maps from this sensor, however the missing hardware synchronization is an issue. Plus the color cameras are rolling shutter, which is an effect we do not compensate, so for long exposure times and dynamic motions, the performance will decrease significantly.

huabit, 2018-12-04:
Hi, @mfehr:
    Thank you very much for your answer and advice. I've change the rostopic names of zed mini with short form, for example named them as /imu0 and /cam0, and then Maplab can subcribe them. But the VIO outputs of Maplab with zed mini are not good.
    Maybe I have to change the sensors.