ethz-asl/maplab #104: Failing to run maplab_console

K0stIa, 2018-10-15:
rosrun maplab_console maplab_console gives me following output

`➜  /maplab_ws rosrun maplab_console maplab_console           
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
*** Aborted at 1539611895 (unix time) try "date -d @1539611895" if you are using GNU date ***
PC: @     0x7f02a7642428 gsignal
*** SIGABRT (@0xa9d) received by PID 2717 (TID 0x7f02a878db00) from PID 2717; stack trace: ***
    @     0x7f02a76424b0 (unknown)
    @     0x7f02a7642428 gsignal
    @     0x7f02a764402a abort
    @     0x7f02a7c7c84d __gnu_cxx::__verbose_terminate_handler()
    @     0x7f02a7c7a6b6 (unknown)
    @     0x7f02a7c7a701 std::terminate()
    @     0x7f02a7c7a919 __cxa_throw
    @     0x7f02a7c7aebc operator new()
    @     0x7f02a7d0c499 std::__cxx11::basic_string<>::_M_mutate()
    @     0x7f02a7d0d833 std::__cxx11::basic_string<>::_M_append()
    @     0x7f02a818d7b3 std::operator+<>()
    @     0x7f029a00ad05 _ZN7rospack11Rosstackage12getCachePathB5cxx11Ev
    @     0x7f029a00b1b4 rospack::Rosstackage::validateCache()
    @     0x7f029a00cdea rospack::Rosstackage::readCache()
    @     0x7f029a00f88c rospack::Rosstackage::crawl()
    @     0x7f029a020bd4 rospack::rospack_run()
    @     0x7f029a01d017 rospack::ROSPack::run()
    @     0x7f029a01d4fa rospack::ROSPack::run()
    @     0x7f029fbb0fc1 ros::package::command()
    @     0x7f029fbb1221 ros::package::getPath()
    @     0x7f02a65d927b pluginlib::ClassLoader<>::ClassLoader()
    @     0x7f02a65d99d5 boost::make_shared<>()
    @     0x7f02a65c3bc0 image_transport::ImageTransport::ImageTransport()
    @     0x7f02a684b6cb visualization::RVizVisualizationSink::initImpl()
    @     0x7f02a684b97d visualization::RVizVisualizationSink::RVizVisualizationSink()
    @     0x7f02a7f753fd maplab::MapLabConsole::discoverAndInstallPlugins()
    @     0x7f02a7f757d7 maplab::MapLabConsole::MapLabConsole()
    @           0x4026a4 main
    @     0x7f02a762d830 __libc_start_main
    @           0x402749 _start
[2]    2717 abort      rosrun maplab_console maplab_console`

mfehr, 2018-10-15:
Hi @K0stIa,
Please have a look at the checklist that should be visible when opening a new issue.
We need some more information to be able to help you.

K0stIa, 2018-10-15:
Hi @mfehr, thanks for the reply. In fact, the problem was on my side. Sorry for bothering you.

Cheers.

closed 2018-10-15 by K0stIa