ethz-asl/maplab #105: Attach color images to VI map

germanrodriguezlopezdelavega, 2018-10-15:
**TODO:** ADD YOUR QUESTION/PROBLEM DESCRIPTION HERE
Hi,

I need to attach color images to my VI map for a dense reconstruction with pmvs. I am running Rovioli on VIO mode with --map_builder_save_image_as_resources flag set to true. Grayscale images are saved under resource folder, but Is it possible to attach color images instead of grayscale images?
I also tried to use resource importer. I have clone and built march 2018 candidate release, but resource_importer package is not available on my workspace.

Thank you,
---

Before submitting this issue, please consider the following **checklist**:

 - Consult the [FAQ](https://github.com/ethz-asl/maplab/wiki/FAQ) and [Known Issues](https://github.com/ethz-asl/maplab/wiki/Known-Issues) pages.
 - Make sure maplab was installed using the proper [installation instructions](https://github.com/ethz-asl/maplab/wiki/Installation-Ubuntu)
 - For bugs and build problems please add (if applicable):
   - Version of:
     - Operating System (32/64bit, 14.04/16.04/...)
     - Eigen
     - ROS
   - A backtrace (if maplab crashed)
   - Run `catkin config` in your workspace and post it here.
   - Details about the commands/scripts used.
   - Details about special parameter values or manual changes to the system.
   - Pictures/screen recordings
 - For general questions:
   - Please elaborate, we are no mind readers :)

mfehr, 2018-10-15:
Hi @germanrodriguezlopezdelavega

`Is it possible to attach color images instead of grayscale images? `

No, it's currently not possible, the flag only attaches the images that are used for mapping.

`I also tried to use resource importer`

That's the way to go. It is definitely available in the march release branch (Have a look [here](https://github.com/ethz-asl/maplab/tree/pre_release_public/march-2018/tools)). You have to build the package, as it is not a dependency of the maplab metapackage. I just added this to the instructions on the wiki.

germanrodriguezlopezdelavega, 2018-10-16:
Thank you @mfehr, I built resource_importer package and that worked.

Now, when i run resource importer, the following error is reported: 
`F1016 10:50:48.991763 11085 resource-importer.cc:146] This image resource type is currently not supported by this importer! encoding: bayer_rggb8`

How can I solve this?

Thank you,

mfehr, 2018-10-16:
If you follow the error message you will see that we currently only implemented the import function for depth images (TYPE_16C1) and color images (TYPE_8C3) and not all [50+ possible types](http://docs.ros.org/jade/api/sensor_msgs/html/namespacesensor__msgs_1_1image__encodings.html).

The color image format we support is the opencv cv::Mat with 3 unsigned 8 bit channels so you will need to add an additional if case [here](https://github.com/ethz-asl/maplab/blob/pre_release_public/march-2018/tools/resource-importer/role/resource-importer.cc#L145) and convert the image to the correct cv::Mat format (basically the image we store [here](https://github.com/ethz-asl/maplab/blob/pre_release_public/march-2018/tools/resource-importer/role/resource-importer.cc#L133) for the supported color image)