ethz-asl/maplab #106: Using Maplab with a camera without IMU?

DodsGTH94, 2018-10-18:
Hi everyone,

I would like to know if it´s possible to use the Maplab with a camera which has no IMU and which things I would have to change.

Thank you

mfehr, 2018-10-18:
Hi @DodsGTH94 
Short answer: no, please don't :)
Long answer: In my opinion it doesn't make sense to adapt a visual-inertial framework to use vision only, when there are other frameworks that do just that (e.g. ORB-SLAM). There are just too many places in the framework where we assume the presence of the IMU and you would need to find them all and find a different solution.

DodsGTH94, 2018-10-18:
Yeah, that's exactly what I expected.
Thank you, I'll investigate other ways!!

Btw beautiful code.