ethz-asl/maplab #107: Visualize rovioli as a ros topic

neorobo, 2018-10-24:
Does maplab/rovioli already support outputting the rovioli feature tracking visualization image as a ros topic? If not, any suggestions on how I can add this?

schneith, 2018-10-25:
Yes, these visualizations are available; unfortunately they are a bit hidden :)

You can use the following flags to enable these visualizations:
```
-feature_tracker_visualize_feature_tracks
-feature_tracker_visualize_keypoint_matches
-feature_tracker_visualize_keypoints
-feature_tracker_visualize_keypoints_individual_frames
```
The images will then be published on a ros topic.

Also for us it is impossible to remember all these features/flags and we usually do a search over the GFlags to find them. In this case I was using:
```
rosrun rovioli rovioli --help | grep tracker | grep visualize
```

mfehr, 2018-10-25:
There is a [section](https://github.com/ethz-asl/maplab/wiki/Running-ROVIOLI-in-VIO-mode#visualization-of-maplab-feature-tracking-module) in the wiki about this.

neorobo, 2018-11-23:
But those are for BRISK/FREAK features used in maplab no? I'd like to see the ROVIOLI ones that are normally displayed outside of ROS.

mfehr, 2018-11-24:
True. There is a way in rovio to visualize the patches but I'm not too sure if this still works after wrapping ROVIO in maplab (=ROVIOLI). The setting is here:

https://github.com/ethz-asl/maplab/blob/master/applications/rovioli/share/rovio_default_config.info#L78

To publish this as a ros topic just try  to replace the cv::imshow calls with ros publishers, here:

https://github.com/ethz-asl/maplab_rovio/search?q=cv%3A%3Aimshow&unscoped_q=cv%3A%3Aimshow