ethz-asl/maplab #110: Depth sensor for localization

MetaI, 2018-10-30:
Hi maplab team!

I got some doubts, mainly because of my lack of knowledge in this field. So I apologize beforehand.
I calibrated a Lenovo Phab 2 Pro Tango device's camera, more specifically the fisheye camera for mapping and localization. This device's depth sensor is also exposed thanks to TangoRosStreamer app so I also hace the chance of recording the pointcloud too in the mission's bagfile.
Appart from 3D reconstruction, would it be useful or can it be used for localization? I mean wether it would be useful or not for improving localization. If so, how should I incorporate this data into map creation and localization?

dymczykm, 2018-11-29:
Hi @MetaI,
honestly, there's no obvious and direct way to include the dense data in the localization pipeline. If you want to explore some research directions, it might make sense to look into depth for pose refinement (by ICP or similar) or for segment-based localization (see [Segmap](https://github.com/ethz-asl/segmap)). The more straightforward benefits would rather come from using depth in the odometry estimation pipeline, e.g. to initialize feature depth. But we have none of that in the current codebase.

MetaI, 2019-01-03:
Thank you very much @dymczykm for taking your time for pointing me to new ideas :) I'll check those.

closed 2019-01-03 by MetaI