ethz-asl/maplab #111: Data for representing localization paths

MetaI, 2018-10-30:
Hi again Maplab team!

I would to represent recorded missions from a Tango device in a website,
Right now I'm processing localization missions recorded rosbag files and also using them for map refinement.

How can I export recorded paths in order to have enought data to paint them escalated in the website? 

I was reading the CSV export section of the wiki just in case, and it seems, if I understood well, that the missions incoporated into the map can be exported in CSV. 

Is it a good start point? Could you give me a clue about the best path for achieving this purpose?

MetaI, 2018-11-07:
All right I found T_G_I , so I recorded a rosbag file with this topic and converted it to csv. Could you tell me what unit is used to express it? Is it meters?

schneith, 2018-11-07:
Yes, the translation part should be in meters.

MetaI, 2018-11-08:
Also,  I think I don't understand well T_G_I and T_G_M, do I need both to localize myself in a map?, what do they mean? Where I can find documentation about it in order no to do silly questions?