ethz-asl/maplab #113: Trainning BoW

wendaoyuchen, 2018-11-08:
Hi,I am struggling with relocalization with different camera in maplab and I found your suggestions in issue #85(url) , thanks for your ideas. And I have some questions about how the vocabulary works in maplab.
1)  I want to expand the dimensions of projected descriptor from 10 to 20,30 or 40 .If I need to train another vocabulary ? Will it works for improving the relocalization?
2) I found the vocabulary is loaded as below:
`vocabulary_.Load(&in);` 
and the `in` is a string "/inverted_multi_index_quantizer_brisk.dat", so I guess this is the bag of words, but the file located at "~/maplab_ws/devel/share/loopclosure/" is just 629.9KB. It is fully different from the vocabulary in ORB-SLAM and VINS, which are about 100~150MB. Is that you training it with descriptors which is 10 dimension ? Futher, the freak.dat in the same folder is even small, just 60.5KB. 
3) In that folder, there are two another files besides quantizer_brisk.dat and quantizer_freak.dat, it seems like something about projection matrix, I guess it works when processing frame to projected_image as below
` *projected_descriptors =
      projection_matrix.block(
          0, 0, target_dimensions, projection_matrix.cols()) *
      converted_descriptors.block(
          0, 0, projection_matrix.cols(), projected_descriptors->cols());`
it cut the descriptor from 384 to 10, but why dose the cutted descriptor multiply the projection? To convert  the descriptor from binary space to real value space?What dose the projected matrix represent for?

dymczykm, 2018-11-29:
Hi @wendaoyuchen,
1. Yes, you would need to train another projection matrix and inverted-multi-index vocabulary files.
2. The files are completely different than in ORB-SLAM or VINS, as we're *not* using bag-of-words, but inverted multi-index. 
3. The matrix is a linear projection matrix obtained using Principal Component Analysis (PCA) that projects the binary descriptors to the real-valued Euclidean Space.

I would suggest to read some relevant papers to understand better the system:
* _Get Out of My Lab: Large-scale, Real-Time Visual-Inertial Localization_ http://www.roboticsproceedings.org/rss11/p37.pdf
* _Placeless Place-Recognition_ https://ieeexplore.ieee.org/document/7035839
* _Efficient Descriptor Learning for Large Scale Localization_ https://ieeexplore.ieee.org/abstract/document/7989359

robot-mh, 2020-05-27:
How to train projection matrix and inverted-multi-index vocabulary files？Any tutorials？ Thank you.