ethz-asl/maplab #115: Publishing covariance matrix after update step

maartenvlaswinkel, 2018-11-13:
Hi,

Is it possible to publish the covariance matrix after the update step as a ROS topic? Or what would be the best way to publish this information.

I know that the update step is done in ```maplab_rovio/lightweight_filtering/include/Update.hpp``` with the state as ```filterState.state_``` and covariance as ```filterState.cov_```, if I am not mistaken. However, I was not able to find the 'route' of the data to the publishing of the topics in ```rovioli/scr/data-publisher-flow.cc```. Is there somebody who could clarify this, as it makes sense to publish the covariance also in this part of the code.

To give some background information. I am using a micro-drone with several sensors. To also use the other sensor that, I am running a separate Kalman filter on the micro-drone. To improve the estimate, of mainly the global position, I want to fuse the estimate of ROVIOLI as a measurement in this Kalman filter.

dymczykm, 2018-11-13:
@maartenvlaswinkel so we're using the RovioInterface abstract class to access the state of the filter. 

An example to access state would be to look into our health monitor we have implemented in release candidates branches, see here for example:
https://github.com/ethz-asl/maplab/blob/pre_release_public/july-2018/applications/rovioli/src/rovio-flow.cc#L175
The health monitor object is getting access to the state and it can read feature covariances. I understand you would need something similar and then a message to data-publisher-flow to publish it as ROS. Does that help? Do you have any further questions?

mfehr, 2018-11-13:
Probably a nice way would be to extend the datastructure of [RovioEstimate](https://github.com/ethz-asl/maplab/blob/master/applications/rovioli/include/rovioli/rovio-estimate.h) to also include the covariance, then you simply need to extend the callback that publishes the RovioEstimate in the [data-publisher-flow](https://github.com/ethz-asl/maplab/blob/master/applications/rovioli/src/data-publisher-flow.cc#L86)

maartenvlaswinkel, 2018-11-14:
Hi @dymczykm and @mfehr,

I think that the method described by @mfehr a nice way to extend the data being published by ROVIOLI. Thanks for the tips.

closed 2018-11-14 by maartenvlaswinkel

neorobo, 2019-02-07:
Hi @maartenvlaswinkel , did you end up having success doing this?

maartenvlaswinkel, 2019-02-07:
Hi @neorobo, I used the method proposed by @mfehr. Sadly I cannot give you more guidance, because I haven't been using ROVIOLI since the end of December and don't remember the changes I made.