ethz-asl/maplab #116: how find the interface of Binocular vision to maplab？

Xuleileilei, 2018-11-23:
Where is the interface between binocular vision and Maplab? I want to use SML to make SLAM mapping through maplab.
Please help me ！
THANKS!

Xuleileilei, 2018-11-23:
I want to use  binocular vision to make SLAM mapping through maplab.

Xuleileilei, 2018-11-23:
Can I import the video format file directly into maplab?

dymczykm, 2018-11-29:
Hi @Xuleileilei,
what exactly do you mean by SML and binocular vision? We generally support the visual-inertial sensing modalities, which needs you need both a camera and an IMU sensor to run maplab.

Xuleileilei, 2018-11-29:
thank you for answering my questions，SML Is the word I misspelled，I am sorry for the meaning of the problem I expressed，now my question has been answered.

closed 2018-11-29 by mfehr