ethz-asl/maplab #117: Performance on ADVIO datasets

zhukovv, 2018-11-28:
Hello!

Is there a chance to make Rovioli work with iPhone data, like ADVIO?
Or it's impossible because of rolling shutter cameras, cheap IMU etc?

In the ADVIO [paper](https://arxiv.org/pdf/1807.09828.pdf) authors say:
"There are many sequences where both methods diverge completely ... On the other hand, there are also sequences where they work reasonably well" 
and also
"ROVIO had problems with long-term occlusions and disagreements between visual and inertial data. Also, in Figure 5e it has clearly inaccurate scale—most likely due to the not modeled scale bias in the accelerations". 

Could it be fixed with maplab console post-processing and then running in localization mode?

I tried original Rovio (not Maplab version) with ADVIO data and my own iPad data with fixed focus, and the tracking seems to work (when the images are not blurred), but the pose drift upwards - though I'm not sure about my config and axes alignment (comparing to Euroc).

Thank you very much!

---

Before submitting this issue, please consider the following **checklist**:

 - Consult the [FAQ](https://github.com/ethz-asl/maplab/wiki/FAQ) and [Known Issues](https://github.com/ethz-asl/maplab/wiki/Known-Issues) pages.
 - Make sure maplab was installed using the proper [installation instructions](https://github.com/ethz-asl/maplab/wiki/Installation-Ubuntu)
 - For bugs and build problems please add (if applicable):
   - Version of:
     - Operating System (32/64bit, 14.04/16.04/...)
     - Eigen
     - ROS
   - A backtrace (if maplab crashed)
   - Run `catkin config` in your workspace and post it here.
   - Details about the commands/scripts used.
   - Details about special parameter values or manual changes to the system.
   - Pictures/screen recordings
 - For general questions:
   - Please elaborate, we are no mind readers :)

adamek727, 2018-11-28:
I am currently working on a very similar topic, and yes. ADVIO-like data works quite well with ROVIOLI. The output map is not so high quality as for example the Euroc-based one but still it works.

zhukovv, 2018-11-29:
@adamek727 
oh, cool, thank you!
so you've converted the datasets into rosbags and put the calibration data into the config file, and that's all?

adamek727, 2018-11-29:
Actually I am not using the ADVIO data, but my own records, but yes. Just recorded the camera, gyro and acc, put everything into bag, make calibration and it works.

zhukovv, 2018-11-29:
@adamek727 
sounds great, thank you!

adamek727, 2018-11-29:
Actually there is one important think to remember. The Kalibr and Maplab are using different direction of gyro data orientation compared to iphone. This cause the bed results of calibration. To fix it you have to put minus sign to gyro OR acc data when you recording it. This is going to fix the problem and the calibration will give you good results.

zhukovv, 2018-11-30:
@adamek727 

oh, thank you, that's really helpful!
yes, I've noticed this kind of issues, but didn't know how to fix it properly. I put minus to both acc and gyro... So just to one of them, but to all three axes, right?

adamek727, 2018-12-04:
Right

scortes48, 2019-04-05:
Hi. 
about the ADVIO data.
There is a bug in the timing of the frames, The first frame should have the timestamp 0.0. So there is a roughly 300 ms desync.

The fix is:
t_frame=t_frame-t_frame[0]

The acceleration is already multiplied by -1. I have run ROVIOLI on the raw data with the timing fix and it works, sometimes it diverges on highly dynamic scenes and it is unable to recover. In the static scenes it has some drift but it works.