ethz-asl/maplab #118: Why message_flow instead of ROS topics?

GoosebumpsFactory, 2019-01-05:
We were curious as to why does maplap define it's own publisher/subscriber based message passing mechanism in `maplap/common/message_flow` instead of using similar functionality available in ROS (i.e., topics)? 

---

Before submitting this issue, please consider the following **checklist**:

 - Consult the [FAQ](https://github.com/ethz-asl/maplab/wiki/FAQ) and [Known Issues](https://github.com/ethz-asl/maplab/wiki/Known-Issues) pages.
 - Make sure maplab was installed using the proper [installation instructions](https://github.com/ethz-asl/maplab/wiki/Installation-Ubuntu)
 - For bugs and build problems please add (if applicable):
   - Version of:
     - Operating System (32/64bit, 14.04/16.04/...)
     - Eigen
     - ROS
   - A backtrace (if maplab crashed)
   - Run `catkin config` in your workspace and post it here.
   - Details about the commands/scripts used.
   - Details about special parameter values or manual changes to the system.
   - Pictures/screen recordings
 - For general questions:
   - Please elaborate, we are no mind readers :)

smauq, 2019-01-11:
It was done both for performance reasons to avoid having multiple in-memory copies of the same message as well as to have a guarantee that the messages will be delivered in the exact same order as they were sent also sometimes across different flow topics (see exclusivity groups in the code).

For example ROS nodelets handle the efficiency part of not using multiple copies, but don't have guarantees on the ordering of messages at arrival.

mfehr, 2019-01-11:
Maybe a small addition to what @smauq said, another reason is simply that in maplab we tried to avoid having any ROS components on the algorithmic level of the system. We use ROS only for visualization and data input, and the latter we tried to keep flexible enough so someone could interface it with another type of data source.

GoosebumpsFactory, 2019-01-11:
Thanks for that explanation. That makes sense.

closed 2019-01-11 by GoosebumpsFactory