ethz-asl/maplab #119: IMU measurements problem when I try to run maplab with MyntEye

bxc237, 2019-01-09:
Hi,
When I tried to run maplab with MyntEye camera, it can not run as stable as running with ZR300. Typically, after running about 40 mins, the number of IMU_MEASUREMENTS queue elements would go up to 3000 with the complaint about IMU measurements lag behind (fig.1), then maplab would die, or the number of VIO_UPDATES would go up to 8500 with the complaint about too few IMU measurements (fig.2), then maplab would die. 
[image]
fig.1
[image]
fig.2
This issue should be relevant to https://github.com/slightech/MYNT-EYE-S-SDK/issues/25 and I also see that @mfehr has the edited version of the MYNT-EYE-S-SDK. However, if I modified the SDK as @mfehr did in https://github.com/ethz-asl/MYNT-EYE-S-SDK/commit/03dae5c458254dc5c2f23b7bf9877979be790e80, the problem would present much earlier so the original MYNT-EYE-S-SDK actually gave me a better result. Did maplab team already find a way to run maplab stably with MyntEye?  Should I try to patch the cuckoo_time_translator in the SDK somewhere as you guys did in ZR300 driver to improve the performance? Thank you!

mfehr, 2019-01-11:
Hi @bxc237,
Yes, we obtained a MYNT-EYE-S sensor and tried to build VI-maps. Before doing that I analyzed the IMU messages to detect drops or problems with the timing, as in our experience this is something that a lot of VI-sensors struggle with and later leads to problems. I reported it in the issue you mentioned. After increasing the queue size (devel/mfehr branch) I would say it was working pretty well. However, I never ran it for as long as you did, i.e. I never did any long-term mapping/large scale tests. 

In fig. 1 ROVIO seems to lack behind, how did the estimator result look at that point? One cause might be that the initial calibration is so bad ROVIO cannot correct it and is slowed down. Another reason is that it might simply have a large drop in IMU messages.

In fig. 2 the error you get from ROVIOLI message is pretty clear, but what's causing it I'm not sure. Might be the same issue, that IMU messages are dropping or being delayed.

My money would be on ROS dropping IMU measurements, however we also cannot exclude that there is something weird happening in the message flow system or the camera/imu synchronizer component, since we haven't tested ROVIOLI on MYNTEYE for very long datasets.

Here is what I'd do:

* checkout my calibration ([here](https://gist.github.com/mfehr/002077dd305b2b58103bd6d020432fb7)) and compare it to yours to see if there are some fundamental differences that might be causing this. Consider re-calibrating or using the calibration refinement tutorial in the maplab wiki.
* record a rosbag that shows this problem, so you can analyze the imu messages and general message timings. I modified [an existing script that restamps the rosbag](https://gist.github.com/mfehr/3d6cc1cf4ba293968d999d62493c56a7) to also visualize the drops a bit, you can give it a try, but be aware it's just a quick hack, I'd be surprised if it's plug and play.
 * if nothing works, we'd be interested in getting some data to reproduce this issue.

bxc237, 2019-01-15:
Hi @mfehr,
After further investigation, I found that this problem might be irrelevant to the IMU messages. Those warnings came out when the NUC's 6G memory was almost fully used after running for a long time, which might cause maplab/rovioli cannot process those queue data on time. I suspect that rovioli is writing the VI map data to the RAM all the time even we do not want to generate the VI map after the long run. If I am right about it, is there any way to prevent rovioli recording the VI map data to the RAM?

bxc237, 2019-01-16:
Hi, @schneith with the flag set to true, rovioli would eat all memory after 2 hours in VIO mode or 40 mins in localization mode. The problem seems to be solved now with the flag set to false. Close issue...

bxc237, 2019-03-01:
Hi, @schneith, I got one more question regarding this camera and safeWarningTime. I modified maplab a little by adding wheel encoder data to the rovio flow and leverage the velocityUpdate interface in the original rovio to solve the scaling problem. This modification works perfectly when I use ZR300 with velocity update. But when I implement rovioli with MyneEye and velocity data, it would complain about "[FilterBase::addUpdateMeas] Warning: included measurements at time 7.299629212 before safeTime 7.422387182",  then velocity update would be rejected. I have tried to set the fisheye and imu publishing frequencies equal to ZR300's without success. Do you know where is the problem since I saw this warning was added by you? Implementing rovioli with only MyntEye works well without complaint, rovioli can even do localization in the map built by ZR300. Thank you.

neorobo, 2019-03-12:
Hey @bxc237 this sounds very cool, is it possible that you could share your modification? I am in the process of adding UWB range measurements to ROVIO and I think seeing how you've integrated the velocity updates would be helpful (I might also make use of that).

bxc237, 2019-03-13:
Hi @neorobo, the modifications are simple, but those modifications are separated in different files. You need to register velocity callback in datasource-flow.h (like the way implemented by the registerImageCallback)...convert ros velocity to maplab velocity in ros-helpers.h...add ros velocitycallback in datasource-rostopic.cc (like the image_callback) ...registerSubscriber for the velocity measurements and call processVelocityUpdate already provided by RovioInterfaceImplInl.hpp in rovio-flow.cc...define the ros velocity topic name in vio-common/rostopic-settings.cc...Anyway, just follow the original way that maplab feeds the ros camera datasource to the rovioli data flow.

closed 2019-03-13 by bxc237

neorobo, 2019-03-22:
Hi @bxc237 I'm using the pozyx UWB sensor, not decawave. I can't say a lot about my application, but it will involve entering a large space with few obstructions and a single entry point. My plan is to try use a single UWB range measurement with a beacon set at the entry point together with ROVIO's estimate to further constrain the estimate (of course the single range measurement won't observe the position, but should help constrain the drift in ROVIO).

niekez, 2020-01-22:
I'm using the develop branch with the MYNT EYE S1 camera. In localization mode, ROVIOLI is filling the memory even with the above flag set, my run script is as follows:

```sh
#!/usr/bin/env bash

source ../../maplab_dev/maplab_ws/devel/setup.bash 

rosrun rovioli rovioli \
    --alsologtostderr=1 \
    --v=2 \
    --datasource_type="rostopic" \
    --sensor_calibration_file="../calibration/imu-cam/maplab/sensors_mynteyes_v2.yaml" \
    --vio_localization_map_folder="maps/loc_maps/opt_maps_dev_merged_summary_sparse" \
    --rovioli_run_map_builder=false \
    --rovioli_enable_health_checking=true
```
I'm getting the following errors but don't think they are related:
```console
W0122 18:25:46.953943 30786 imu-measurements-buffer.cc:172] Timeout reached while trying to get the requested IMU data. Requested range: 1579713946s 305687428ns to 1579713946s 455567360ns.
W0122 18:25:46.954144 30786 imu-measurements-buffer.cc:179] The relevant IMU data is not yet available.
W0122 18:25:46.954196 30786 imu-camera-synchronizer.cc:188] NFrame-IMU synchronization timeout. IMU measurements lag behind. Dropping this nframe.
W0122 18:25:47.453649 30786 imu-measurements-buffer.cc:172] Timeout reached while trying to get the requested IMU data. Requested range: 1579713946s 805297375ns to 1579713946s 955177307ns.
W0122 18:25:47.454064 30786 imu-measurements-buffer.cc:179] The relevant IMU data is not yet available.
W0122 18:25:47.454180 30786 imu-camera-synchronizer.cc:188] NFrame-IMU synchronization timeout. IMU measurements lag behind. Dropping this nframe.
```