ethz-asl/maplab #120: Using MAPLAB without using ROVIOLI

str-ygomi, 2019-01-10:
Hi,

I am trying to load output data from a different SLAM framework into Maplab. The data I currently have is as follows:
1. Keyframes (poses, IMU bias, timestamp)
2. Map-points (3d location)
3. Observations (Descriptor, 2d image point, keyframe ID, map-point ID)

My first question is, is the above data (provided it is loaded correctly) enough to run some of the MAPLAB algorithms (I'm trying to get pose relaxation working first)? I am using the code from VIMapGenerator::generateMap(), altered to use my above data (loading kf poses as vertex poses, as though they are IMU measurements). So far results are not very encouraging: triangulation results in very strange point positions (far below ground, or in strange arc shapes), vision-only loop closure finds no loops (even with two identical maps). Is there fundamental data I need to add in, or is it some problem with how I have loaded/associated the data?

And one last question: in this context, can I understand T_G_M to be a transformation between the coordinate system of my saved output data and the coordinate system MAPLAB uses?

Thanks!

str-ygomi, 2019-01-11:
Further:
I don't understand why the loop closure algorithm can't find any matches, even when I load descriptors from one of the datasets provided by the wiki; even if I just set all the descriptors to the same value there are no matches (when surely the distance between all of them is 0). What step am I missing out..?

mfehr, 2019-01-11:
Hi @str-ygomi,
Without knowing more details about what data (Descriptor type?, is there IMU data for the edges? other map format?) you are feeding into the VI-map and how you do it (do you have some code to look at?) it's hard to tell why it's not working.

But here are some suggestions/hints from my side and from @schneith :

 * Execute a consistency check on the VIMap after building it (the function is [here](https://github.com/ethz-asl/maplab/blob/88d8587958c6902e7304a5ef7d88e867b30924b3/map-structure/vi-map/include/vi-map/check-map-consistency.h))
 * save/load the map, if this works without any check failures, visualize it in the console (attach the screenshot here)
 * Check if rotation convention is correct. 
 * The list of things you import does not contain any camera calibration
 * The descriptors are binaries, so there are some things that can go wrong depending on how you write them to files and when moving to and from different PCs. 
 * T_G_M is used to anchor different datasets/session in a common coordinate frame. But of course you can use it to transform a map that is following a different coordinate system convention (e.g. not gravity aligned) to the standard Maplab/ros coordinate system. I guess as long as the data is also expressed in a right-handed coordinate system this should work.
 * what exactly do you mean by 0 matches when doing loop closure? 0 loop closures (as in final loops found) is to be expected if your map geometry is broken, since RANSAC will reject everything, this will happen independent of how good the descriptor matches are. What you could check is raw descriptor matches after the kNN/inverted-multi-index lookup, i.e. before RANSAC. But to be honest as long as the map looks broken right at the beginning after triangulation, there is no point in trying to do loop closure.

I hope this helps.

str-ygomi, 2019-01-14:
Hi @mfehr , thank you very much for your response!

To answer those initial questions:
- The descriptor being used is a custom one made for handling different weather conditions (it's not public). The original descriptors are 48x1 floats which I am converting to a 48x1 unsigned chars (normalise elements, multiply by 255), then loading them into a corresponding vertex.
- I'm not actually loading any IMU data directly; only final pose estimates from this other SLAM system found using IMU, GPS, camera. I am putting these poses in as T_G_I when I create a vertex. I guess I am pretending it's an IMU pose estimate. The edges are just added between these poses via VIMapGenerator::generateMap()
- Forgot to mention the first time, but yes I am also loading camera configuration, eg fu fv cx cy, distortion and TCB

So, firstly I did all those consistency checks (just to be safe) and did find a small problem with landmark association, which is now fixed. The rotation convention was also definitely wrong, but I think I have also fixed that now too (will attach a photo). Loading and saving works fine. As of now, the triangulation is better, although only about 15-20% of points survive. Before and after:
[image]
[image]

Earlier I assumed that a vision only loop closure would find loops regardless of the errors in pose, so I guess I was wrong about that. As of now, after fixing the pose mistakes, it seems as though matches are found (frame_matches is not empty in loop-detector-node.cc, "Successfully loopclosed x vertices"), but RANSAC rejects everything, so maybe the map is still broken somehow..!

Any further advice would be greatly appreciated, thank you for the time spent!!

mfehr, 2019-01-14:
Some more suggestions:
 * your landmarks are way too sparse, probably the "rtl" step is filtering a lot of them. Have a look at the command and play with the thresholds a bit to get more. It's also hard to tell from the few landmark if they represent some meaningful geometry. An easy check is to run visualize_sequentially to see if all landmarks associy with a vertex are meaningfully distributed and roughly in the camera frustrum.
 * I recommend that you have a closer look at the paper and the publication describing the loop closure. What maplab does is it projects the **binary** descriptor to a subspace using a projectionmatrix trained for brisk. Your descriptor happens to have the right length I think but the projection produces meaningless results which probably leads to random matches which cannot pass verification because, well, they are probably random. You can try to train or come up with your own projection matrix or you have to work around the projection and directly do a knn lookup on the float vector. We do use either a kd tree or an inverted multi index, but again, it's on the projected descriptors (which are floats), so you would need to bypass the projection and directly use your floats.
 * Provided you find a way to modify loop closure to give you meaningful constraints, if you plan on running bundle adjustment later you cannot use optvi if you don't have IMU data. You can try optv (vision only) but considering how few landmarks you have I'm not sure how well this works. Maybe give it a try now and check if it does anything meaniny, if your initial guesses are already good it might stay stable which would be a good quality check, but it might just tear your map to bits...
 * if you do end up implementing an extension to loop closure that can deal with non binary descriptors and without projection, this could be a pretty ccol additional tool, so we'd be happy to hear how it goes :)

closed 2019-05-14 by mfehr