ethz-asl/maplab #124: Proper way to run gtests

aharmat17, 2019-01-30:
Hello,

What is the proper way to run gtest executables? I did the following for the map_resources package:

`catkin build --make-args tests -- map_resources`
`rosrun map_resources test_resource_map`

Which gives me a test failure:
`[==========] Running 14 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 14 tests from ResourceMapTest
[ RUN      ] ResourceMapTest.TestResourceMapAddResourceDefaultFolder
F0130 14:13:13.862169 30900 resources-test.h:78] Check failed: ((depth_cvmat_resource_.type()) & ((1 << 3)*512 - 1)) == 2 (0 vs. 2) 
*** Check failure stack trace: ***
    @     0x7fadefcb826d  google::LogMessage::Fail()
    @     0x7fadefcba083  google::LogMessage::SendToLog()
    @     0x7fadefcb7dfb  google::LogMessage::Flush()
    @     0x7fadefcbaa6e  google::LogMessageFatal::~LogMessageFatal()
    @           0x43eb97  backend::ResourceTest::loadTestResources()
    @           0x450919  backend::ResourceTest::createResourceTemplates()
    @           0x42045e  backend::ResourceMapTest_TestResourceMapAddResourceDefaultFolder_Test::TestBody()
    @     0x7fadf16c7243  testing::internal::HandleExceptionsInMethodIfSupported<>()
    @     0x7fadf16beb5a  testing::Test::Run()
    @     0x7fadf16beca8  testing::TestInfo::Run()
    @     0x7fadf16bedb5  testing::TestCase::Run()
    @     0x7fadf16c0a6f  testing::internal::UnitTestImpl::RunAllTests()
    @     0x7fadf16c0da9  testing::UnitTest::Run()
    @           0x41b881  main
    @     0x7fadef36a830  __libc_start_main
    @           0x41c929  _start
Aborted (core dumped)`

It seems the problem is that map_resources_test_data.tar.gz is not unzipped, which should be done by
`add_custom_command(TARGET ${PROJECT_TEST_DATA}
COMMAND rm -rf "./${PROJECT_TEST_DATA}/*" && tar -xvzf ${MAPLAB_TEST_DATA_DIR}/${PROJECT_TEST_DATA}/${PROJECT_TEST_DATA}.tar.gz)` in the map-resources CMakeLists.txt. 

So am I building the tests incorrectly? I'm guessing there's a simple solution to this problem.

Thank you,

Adam

mfehr, 2019-01-31:
There are a couple of different ways how to build/run unit tests. I usually do:
```bash
# build package and its deps
catkin build my_package
# build and run unit tests just for this package
catkin run_tests my_package --no-deps 
```

aharmat17, 2019-01-31:
Thank you, that worked!

closed 2019-01-31 by aharmat17

IoannisKaragiannis, 2020-03-04:
Hi,

I have a relevant question so I'm posting it here instead of opening a new issue.

I am sitting on the master branch of both maplab and the dependencies and I have successfully built your project. Before proceeding with your tutorial I would like to run some unit tests. When I run the tests according to your recommendation
```shell
catkin run_tests maplab --no-deps
```
all the tests pass successfully. However, if I run
```shell
catkin run_tests maplab
```
the test fails in the `cv_bridge` package which of course is a dependency.

1) Have you developed unit tests for the dependencies as well or only for maplab itself?

2) Since maplab seems to be using lots of the dependent packages, are my tests still meaningful if I exclude the dependencies?

Thanks in advance for your time.