ethz-asl/maplab #125: Building maps sequentially during the same mission

viktortuul, 2019-01-31:
Hi, 

I am investigating the possibilities of building maps sequentially during the same mission using VIO mode. That is, instead of building the map when ROVIOLI in VIO mode is shut down, actually do this e.g. every 20 seconds. 

I’ve tried to sequentially execute the following code: 
`if (!save_map_folder.empty()) { 
rovio_localization_node.saveMapAndOptionallyOptimize( 
save_map_folder, FLAGS_overwrite_existing_map, 
FLAGS_optimize_map_to_localization_map); 
} 
`
in **rovioli-app.cc**

which builds the map. 

I have also tried with resetting the MapBuilder in **rovioli-node.cc** every time a map is built, using the code below, without any further success.

`void RovioliNode::resetMapBuilder() {
    map_builder_flow_.reset(new MapBuilderFlow(camera_system, std::move(maplab_imu_sensor), save_map_folder));
    map_builder_flow_->attachToMessageFlow(flow); 
}`

However, each built map is a copy of the first one (I guess that the map_builder_flow stops to store landmarks etc. after executing that code) 

The reason why I am investigating this is to enable “semi-online” SLAM combining VIO and Localization mode for re-localization. Furthermore, the goal is to use this as a map-fusion functionality using multiple drones. 

Is it known if this has this been looked into before or are there better ways to go when trying to implement at least a “semi-online” SLAM/map fusion solution using this framework? 

Thanks.

viktortuul, 2019-02-01:
With that said, I am also looking into enabling ROVIOLI in Localization mode to change the currently used map, online, without having to close and re-open the localizer.

An ugly solution would be to just close and re-open the applications when a new map is to be built/read through a command-line script. But that will probably affect the robustness of the state estimate, as everything is reset (if not accounted for in the initialization file for example). Thus I am looking into a more elegant and stable solution.

I have already created a script that automatically performs commands in the maplab console for e.g. map fusion. So the main objectives now are to sequentially build maps in VIO and automatically read new maps in Localization mode without having to close and re-open the applications, as previously stated. 

I  will appreciate any ideas and feedback.

viktortuul, 2019-02-13:
An update:

I have managed to fix a sequential map builder which creates maps at a fixed time interval without having to shut down the ROVIOLI application, as well as an automatic map optimizer (through the maplab_console). 

I am however still struggling with loading a new map in Localization mode without having to restart the application. 

I have tried with resetting the `localizer_flow ` with a new map as input, by creating the following function in `rovioli-node.cc`:
```
void RovioliNode::initializeNewMapOnline(
const std::string& save_map_folder_, 
const summary_map::LocalizationSummaryMap* const localization_map_, 
message_flow::MessageFlow* flow_) {

        constexpr bool kVisualizeLocalization = true;
        localizer_flow_.reset(new LocalizerFlow(*localization_map_, kVisualizeLocalization));
        localizer_flow_->attachToMessageFlow(flow_);
}
```

However, I receive an error message saying the following:
 `
F0213 10:45:39.643873 17167 localization-summary-map.cc:161] Check failed: it != landmark_id_to_landmark_index_.end() Landmark 0000..9176 is not in localization summary map.
 `

I have checked everything I can think of, and I have checked that  `localization_map_ ` is in fact a valid map.

Might there be anything obvious that I am missing?

mfehr, 2019-02-15:
Hi @viktortuul,
we actually had to do something similar for a project, i.e. swapping it during runtime it. When you do this there are a couple of things you need to be aware:
 * the localization map and all the things depending on it  need to be locked during the exchange to prevent issues with multithreading
 * inside the localizer class there are several things that need to be reinitialized when replacing the localization maps or you will see bugs like the one you have now. Check the constructor of the Localizer class to find out what needs to be replaced along with it. I think it's some cache/lookup and the loop closure node itself. You might need to adapt e few interfaces and change the localizer member variables to unique ptrs.

viktortuul, 2019-02-15:
Hi @mfehr,
Thank you for the reply and great insights!

Regarding the first bullet point,  when locking the localization map and the things depending on it, is there a "smart" way of doing that? Or is it by using mutex object locks on all the concerned objects during the period the map is being exchanged?

Thanks!

mfehr, 2019-02-18:
I think we just had one mutex that was responsible for locking localization map, cache and loop closure node when it is exchanged and would also block the localizeNFrame call such that during the exchange no localization can be executed.

robot-mh, 2020-02-06:
Whether replace ROVIO with ORB-SLAM2 could solve this question？