ethz-asl/maplab #126: ERROR: unknown command line flag 'imu_parameters_rovio'

gakkiHp, 2019-02-17:
ERROR: unknown command line flag 'imu_parameters_rovio' 

I get the error when running "rosrun rovioli tutorial_euroc save_folder MH_01_easy.bag", what should i  do?

mfehr, 2019-02-17:
I assume you are using a release branch, they include some changes in the interface. Please read the PR description, it should explain the changes.

gakkiHp, 2019-02-17:
> I'm assuming you are using a release branch, they include some changes in the interface. Please read the PR description, it should explain the changes.

You are right, I am in tx2 maplab installation, bugs in the installation process is according to https://github.com/ethz-asl/maplab/issues/7, here to solve. This problem then arises when the command is run. I am just a novice, so I would like to ask you what the PR description is. Where can I see it?

mfehr, 2019-02-17:
The release branches have a pull request here on the website associated with them with a description.

gakkiHp, 2019-02-17:
> The release branches have a pull request here on the website associated with them with a description.

I think I have found it. It should be modified as per what is said here. 
[http://a2.qpic.cn/psb?/V10dYaKO1BD0DF/mn2lXaXkpvA2B15qndCBaLS7q7BFBphmwYL9A3Tql1w!/m/dLkAAAAAAAAAnull&bo=mwL1AQAAAAADB08!&rf=photolist&t=5](url)
I will try it. However, I have to reflect a situation: after I installed the file on tx2, I thought I was missing the file imu-sigmas-rovio.yaml after running the command to report an error, so I added one in the share, but it still reported the error, I would like to ask whether there was no file imu-sigmas-rovio.yaml in the Release branches

gakkiHp, 2019-02-17:
> The release branches have a pull request here on the website associated with them with a description.

I think I have found it. It should be modified as per what is said here. 
**ROVIOLI - Unified maplab and rovio sigmas**

_Previous version:_

Due to a difference in the convention of the IMU parameters, two separate IMU parameter files for ROVIO and maplab had to be passed to ROVIOLI.

```shell
--imu_parameters_maplab=$IMU_PARAMETERS_MAPLAB
--imu_parameters_rovio=$IMU_PARAMETERS_ROVIO
```

_Update:_

We unified the parameters and it now suffices to simply pass the maplab IMU parameters to ROVIOLI.

_Effects:_

* ROVIOLI can now be launched with just these IMU parameters:
  ```shell
  --imu_parameters_maplab=$IMU_PARAMETERS_MAPLAB
  ```

I will try it. However, I have to reflect a situation: after I installed the file on tx2, I thought I was missing the file imu-sigmas-rovio.yaml after running the command to report that error, so I added one in the share, but it still reported the error, I would like to ask whether there was no file imu-sigmas-rovio.yaml in the Release branches？ Thanks a lot.

gakkiHp, 2019-02-18:
@Marius Fehr，Thanks a lot, I just delete --imu_parameters_rovio=$IMU_PARAMETERS_ROVIO and  it works.

gakkiHp, 2019-02-18:
@Fehr，Thanks a lot, I just delete --imu_parameters_rovio=$IMU_PARAMETERS_ROVIO and  it works.

gakkiHp, 2019-02-18:
@mfehr ，Thanks a lot, I just delete --imu_parameters_rovio=$IMU_PARAMETERS_ROVIO and it works.

closed 2019-02-18 by gakkiHp

mfehr, 2019-02-18:
if you scroll a bit further in the PR description, you can see that we replaced the rovio flag with a new flag in case you want to input different IMU sigmas for ROVIO and maplab:

[image]