ethz-asl/maplab #129: Kalibr IMU Camera Config Conversion

gakkiHp, 2019-02-27:
Kalibr IMU Camera Config Conversion

To run ROVIOLI, we need three yaml files inside --ncamera_calibration, imu_parameters_maplab, imu_parameters_rovio. As soon as we connect a sensor, we need to calibrate it ourselves. However, execute the command: Rosrun kalibr kalibr_maplab_config --to-ncamera \ --label \ -- CAM can generate three files? I'm new here. Am I doing something wrong？

mfehr, 2019-05-14:
Hi @gakkiHp,
I'm assuming you did a full kalibr calibration first (cam and imu to cam) and then plugged those results into the script above. The ROVIO imu sigmas you need to provide yourself. The reason for the initial release to have two imu param files is that we didn't unify the imu sigma format between ROVIO and maplab. Try using the newer release branches, which have unified this and have a look at the PR description there, it explains how those sigmas relate.

Derpistan, 2019-06-07:
Hi @mfehr ,
I have been working on the maplab project quite sometime and i have been using the ZR300 camera for rovioli.I have had good output using the camera and the default yaml files provided for the ZR300 camera.But when it comes to calibrating your own yaml files i have no such luck. Let me take you through the whole procedure and reason for opting so:
-The reason i want a calibrate a new yaml for an already existing yaml file is so that i can do that for a different camera because zr300 does not provide its imu data in jetson tx2.
-i used Kalibr to get the camchain.yaml file and now the problem arises as to what noise parameters i should use and how to procure them so as to get the camchain_imucamera.yaml file.
-just to check my camchain.yaml file i used the default imu parametres file provided by maplab to generate camchain_imucamera.yaml file.Turns out the output went haywire and i had a very bad result.
-while all this is going on the master release of maplab(where you require 2 imy yaml files,which is all together a different problem to solve),I started my work on pre_release_public_july2018 maplab just in hope of bypassing the 2 imu yaml file problem.I am facing a build error (some benchmark_config problem).
-So in brief i have to solve this calibration problem and preferably using just one imu yaml file because getting noise parameter of imu is another huge problem i am yet to solve.Also why don't the imu yaml file parameters never match the noise parameters provided in the official miu datasheet ,not even close.

 I hope i have made myself as clear as possible.I would appreciate it if you could help me out on these problems  or if you could point out my mistakes or if there is abetter approach to all this.Thanks

mfehr, 2019-06-26:
Hi @Derpistan
First of all, the camera files and imu files in the repo are only there as a reference/example and to process the datasets we have put online. **Always** calibrate your camera even if you have the same sensor, the IMU sigmas you can try to use our files if you have the same IMU type. However, in the past we have seen that some sensors, even though the same IMU was used produce rather different output, depending on how/if the sensor data is was calibrated (e.g. D435i vs T265). That being said, we agree, choosing these IMU sigmas can be very tricky and it's rarely as simple as taking the ones from the manufacturer. There is a nice description on how to get these IMU parmeters in the [Kalibr wiki](https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model).
Regarding having two IMU files, as I explained above, in a newer release branch we unified these, please have a look at the PR descriptions.