ethz-asl/maplab #130: Performance on android data

gakkiHp, 2019-02-27:
Can we use the image-imu date from android mobile phone as the dataset of maplab(Running ROVIOLI)?

I just  wrote the code to get the android phone's image and imu information, and then I wondered if it could be used as a data source for maplab

mfehr, 2019-02-28:
Hi @gakkiHp 

If the timestamps are good, the main issue remaining is that these cameras are rolling shutter. ROVIO(LI) does't model this and neither does maplab, so it might work as long as you move slowly with short exposure times, but it will likely fail otherwise.

gakkiHp, 2019-03-01:
Thank you. I have another question. Do I need to use kalibr to generate the two calibration files of ncamera-euroc.yaml and imu-adis16488.yaml again

schneith, 2019-03-01:
Yes, you will need to calibrate the sensors. You can use kalibr for this, just make sure to enable time-calibration and rolling-shutter mode.

gakkiHp, 2019-03-02:
Thanks a lot, I saw two calibration methods in kalibr：“Calibrate your sensor and obtain a Kalibr camchain yaml file. Check with the Kalibr wiki on how to do that. The relevant tutorials for a VI sensor: [1] camera calibration, [2] camera-IMU calibration”
I don't know which one to use, I think it should be the second one, right? And then split the generated yaml file into three yaml files?

antmusco, 2019-03-07:
@gakkiHp I was able to get Rovioli running on data from a Galaxy S7, though I had to use the NDK in order to get timestamps from the camera and the IMU with sufficient accuracy. A wide-angle lens also helps a bunch.

gakkiHp, 2019-03-08:
Thank you very much for your help. I would like to ask if you can upload your apk source or apk? I want to try it.

antmusco, 2019-03-12:
@gakkiHp Unfortunately I'm unable to provide the apk do to IP concerns, but I might upload an Android project containing just the sensor streaming service in the future

gakkiHp, 2019-03-12:
@antmusco 
Ok, thank you very much for your help.

closed 2019-04-12 by mfehr

araujokth, 2019-04-25:
Hi @gakkiHp @antmusco I was also interested in trying this out. Any chance you could point me to some documentation which I could use to start recording timestamped camera and IMU data from an Android phone? I wanted to try this on a Galaxy S9. 

@antmusco was curious if it was enough to run Kalibr to get a temporal calibration between the data, or did you have to do anything else to get a good synch?

antmusco, 2019-04-27:
@araujokth For image data, a good starting point would be the [Android NDK Camera sample](https://github.com/googlesamples/android-ndk/tree/master/camera). Specifically:

https://github.com/googlesamples/android-ndk/blob/86ceeb248bb20bfffb09330d1a95bcf3e91bb99d/camera/basic/src/main/cpp/image_reader.cpp#L112

Documentation on the AImage data structure can be found [here](https://developer.android.com/ndk/reference/group/media). A useful note: the function call `AImage_getTimestamp(AImage*, int64_t*)` will populate the passed variable with the timestamp (in nanoseconds) with respect to the system clock.

For the IMU data, you can use the ASensorEventQueue API to read accelerometer and gyroscope events. Again, the Android NDK provides a [pretty good sample for accelerometer data](https://github.com/googlesamples/android-ndk/tree/master/sensor-graph), but is pretty straightforward to extend this to support gyroscope data. The actual sensor readings can be polled in this function:

https://github.com/googlesamples/android-ndk/blob/86ceeb248bb20bfffb09330d1a95bcf3e91bb99d/sensor-graph/accelerometer/src/main/cpp/sensorgraph.cpp#L189

Another useful note: `event.timestamp` contains the timestamp of the sample w/r/t the system clock, which is the same one used to sample camera frames. If you follow this route, the timestamps should be consistent with one another, and there should be no need to calibrated the the temporal offset. Calibration via Kalibr worked out of the box using this method, and Rovioli performed pretty well.

I haven't attempted to use the simpler Android SDK (Java) API for retrieving images/imu data, so I can't speak on how well that would perform. However, I have experienced that lazy stamping like the following was insufficient to prevent Rovioli from exploding:
```
// Thread 1 (poll at 30 Hz)
image_stamp = now()
image_data = get_next_image_data()
// Thread 2 (poll at 100Hz)
imu_stamp = now()
imu_data = get_next_imu_data()
```
Hope this helps!

araujokth, 2019-04-29:
@antmusco thanks so much for all the pointers and suggestions! Really super appreciated. I will give it a try!

araujokth, 2019-04-29:
@antmusco just to confirm, which frequency were you getting when sampling the accelerometer and gyro? I tried the sensorgraph application and even though it is specified to run at 100 Hz (SENSOR_REFRESH_RATE_HZ = 100), I get most of the inter-event times to be 2 ms and very seldomly 1 ms. Thanks again!

araujokth, 2019-05-06:
@antmusco I was now fighting store the image, but I am struggling to get access to images that are higher quality than the ones provided by the sample code using the yuvReader_.  I was wondering which format did you settle on, and if you had any tips on going about getting other data formats from the sample code? 

I tried to change the DrawFrame to have a jpgReader_ instead as AImage *image = jpgReader_->GetNextImage(), but unfortunately what is returned its not a valid image. I tried also to go about by trying to triggering TakePhoto() at 30 Hz, but you can only take a couple of images until it returns that you have reaching the maximum buffer size, so I guess thats not the way to go.

Any tips would be great appreciated, and thanks again for the help!

antmusco, 2019-05-06:
@araujokth What do you mean by "not a valid image"? If you modify the ImageReader to use `AIMAGE_FORMAT_JPEG` then it should work as expected:
```
ImageFormat jpegFormat{.width=640, .height=480, .format=AIMAGE_FORMAT_JPEG};
jpegReader_ = new ImageReader(jpegFormat);
```
This worked for us, and we were able to to copy those bytes directly into a [`CompressedImage`](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/CompressedImage.html) message.

As far as the IMU sensor data, we relied on the uncalibrated data since Rovio does online bias estimation. However we didn't try using the Android calibrated data as-is and disabling the Rovioli bias estimation.

araujokth, 2019-05-07:
What I was trying to do was to use the jpgReader_ which is already created here https://github.com/googlesamples/android-ndk/blob/master/camera/basic/src/main/cpp/camera_engine.cpp#L92 But I also tried to create my own new one like you wrote now. 

Then when I call GetNextimage for this reader, I get a nullptr return from GetNextImage here https://github.com/googlesamples/android-ndk/blob/master/camera/basic/src/main/cpp/image_reader.cpp#L116 since status != AMEDIA_OK, which is what I meant by not getting a valid image. 

For sure I am missing something very silly here...

antmusco, 2019-05-07:
There's your problem! You can remove the image callback functionality -- it's not doing anything for you. If you notice [on this line](https://github.com/googlesamples/android-ndk/blob/master/camera/basic/src/main/cpp/image_reader.cpp#L87) the callback initiates a write to file thread only if the image is of jpeg format.

The write function ends with this:

https://github.com/googlesamples/android-ndk/blob/86ceeb248bb20bfffb09330d1a95bcf3e91bb99d/camera/basic/src/main/cpp/image_reader.cpp#L456

So the callback is deleting your jpeg before you can do anything useful with it.

araujokth, 2019-05-08:
Aha I see! And the capture session is maintained as the default or you have to change that?

https://github.com/googlesamples/android-ndk/blob/86ceeb248bb20bfffb09330d1a95bcf3e91bb99d/camera/basic/src/main/cpp/image_reader.cpp#L456

antmusco, 2019-05-08:
@araujokth To be honest I threw away much of the sample application code and built up the streaming application with dedicated capture loops for both the camera and the IMU. I had no use for display or preview rendering (it also kills battery), so none of the "NativeWindow" stuff was useful to me. All you really need to get up and running are the `AImageReader_acquireNextImage(reader, &image)` to get the JPEG image and `AImage_getTimestamp(AImage*, int64_t*)` to get the timestamp, and push that out over the wire as fast as possible.

Keep in mind that although you are achieving 30Hz image update on the device itself, you will still have to push those bytes over the air. All unnecessary code in the application will only further contribute to latency and jitter on the receiving end. Stripping down as much as I could still barely yielded a 20Hz image update and 180Hz IMU update rate.

*Aside*: It might be neat to receive back and image with overlayed tracking data and display that on the screen, but I expect the round-trip latency would be too high to look clean. Might be fun to try in the future :)

As far as streaming, I made the NDK application a ROS node and published the CompressedImage and IMU messages directly. There may be more efficient ways to handle this. I'd like to try a (GStreamer)[https://gstreamer.freedesktop.org/documentation/tutorials/android/a-running-pipeline.html] pipeline in the future, which may boost the image transfer rate. But as of now the ROS node works reliably, and provides the ability to easily send data back to the device.

araujokth, 2019-05-09:
I see. That makes sense. But you still configured a capture session https://developer.android.com/reference/android/hardware/camera2/CameraCaptureSession or? Did you set it as repeating?

Thanks for the additional tips! :)