ethz-asl/maplab #132: Maplab (Rovioli) Performance Tuning in Nvidia Px2

cedros23, 2019-03-01:
Hi everbody,

Thanks once again for the excellent tool. At the moment, we're using july release in nvidia px2 and we have some performance issues. When I read the paper/issues for both rovio and maplab, I end up with the following useful links:
https://github.com/ethz-asl/rovio/issues/27
https://github.com/ethz-asl/rovio/wiki/Configuration

I would like to change the number of features of rovio to the 15, in order to reduce the computation, however, when I change the maplab_dependencies/internal/maplab_rovio cmake list, then recompile everything. Rovioli still tracks 25 features (green rectangles). Would you please advise?

Secondly, I read in the discussions that reducing the queue size would help to cope with lags in real time critical operation. So, I would like to tune that as well, but I cannot find anywhere to update queue size.

Thanks in advance,
Regards,
Ugur

schneith, 2019-03-01:
You can set the number of features here:
https://github.com/ethz-asl/maplab/blob/master/applications/rovioli/src/rovio-factory.cc#L26

Which queues are you referring to? Are you trying to optimize the performance of VIO+localization or just VIO?

cedros23, 2019-03-01:
Hello,

I really appreciate the quick return!

At the moment, we're working on the VIO, what I observe is, in a 4 camera, 1 imu, 1 velocity (odometry) setup, px2 can handle IMU frequencies around 50 hz. 
Initially, I begin with 100 hz IMU, odometry setup then the problem was booming queues in the rovio information screen when the verbose was 2. I observe that the scene in camera and rovioli was different and rovioli was buffering the measurements then lagging behind real time application.

So, I was thinking to limit queueing or buffer size of rovio while expecting performance loss but still better time critical operation.

Thanks again,

cedros23, 2019-03-11:
Dear Schneith,

In case, I was not clear, I refer to queue sizes below if there is a parameter in the code that I can change in order to limit large queueing and lagging.

[image]

thx again

mfehr, 2019-06-26:
Hi @cedros23,
These queues cannot be shortened per se, if one of those queues is exploding it just means that within ROVIOLI there is a bottleneck somewhere that keeps the data from being processed (in time). I recommend first trying to figure out where this bottleneck is, probably ROVIO or the feature tracker. I would try to throttle the images before they enter ROVIO. We already have throttling implemented in the IMU Cam synchronizer, which can be tuned using this flag: `vio_nframe_sync_max_output_frequency_hz`, but this only influences the feature tracker on the maplab side of ROVIOLI, ROVIO still receives every image, I think.