ethz-asl/maplab #133: Monocular Dense Reconstruction

araujokth, 2019-03-04:
Hi, I was wondering if anyone had looked into integrating monocular dense reconstruction with Maplab or have any thoughts/recommendations of a framework to integrate with Maplab, for example REMODE https://github.com/uzh-rpg/rpg_open_remode?

Thanks for the help!

mfehr, 2019-03-05:
Hi @araujokth 
We originally had a basic mono dense reconstruction pipeline integrated, but we were unable to open-source it along with maplab because of the license. It was looking for suitable pairs of images (suitable = ideal baseline for the avg scene depth of the landmarks) in the pose-graph on which to use either polar or planar rectification and then used the standard ocv block matcher on these images. We haven't looked into anything other than that from our side, but I'm also curious to know if someone else tried and what they used.

araujokth, 2019-04-08:
Thanks a lot for the reply @mfehr! I will have a look at your suggestions!

mfehr, 2019-04-12:
@araujokth  I will close this issue now, but if you are actually pursuing this integration of a monocular dense reconstruction pipeline into maplab (e.g. as a new plugin) we'd be happy to hear how it went :)

closed 2019-04-12 by mfehr