ethz-asl/maplab #135: bgr8 Error when running maplab

gakkiHp, 2019-03-07:
The bag I used is of type bgr8, but maplab has been reporting errors in ros-helpers.h:48, telling me that the bgr8 type is wrong and should be modified, but I noticed that it has the toCvShare function [maplab/applications/rovioli/include/rovioli /ros-helpers.h] Modified the encoding type and why it reported an error. Later I added cvtColor(src,dst,CV_BGR2GRAY) before the toCvShare function, but it still reported the same error, I just wrote some garbled, but it also reported the same error. So is it necessary to compile after modifying the .h file? What do I need to do to compile it?

Inline vio::ImageMeasurement::Ptr convertRosImageToMaplabImage(
    Const sensor_msgs::ImageConstPtr& image_message, size_t camera_idx) {
  CHECK(image_message);
  Cv_bridge::CvImageConstPtr cv_ptr;
  Try {
    // Convert the image to MONO8 if necessary.
    Cv_ptr = cv_bridge::toCvShare(
        Image_message, sensor_msgs::image_encodings::MONO8);
  } catch (const cv_bridge::Exception& e) { // NOLINT
    LOG(FATAL) << "cv_bridge exception: " << e.what();
  }
  CHECK(cv_ptr);

mfehr, 2019-03-07:
Hi @gakkiHp 
We are aware of this issue. The MONO8 fix works if the input types like BGR8, RGB8 or MONO8, but we have quite a bit of data that is of type TYPE_8UC1 and TYPE_8UC3, which cannot be automatically converted to MONO8 by the cv_bridge AFAIK, so one would need to add some more cases to cover our most common datasets/sensors, and we simply haven't gotten around to do it, since it is a minor nuisance at best. Yes you need to recompile the rovioli package.

gakkiHp, 2019-03-07:
Thank you very much for your reply, so I only need to add the function cvtColor(src,dst,CV_BGR2GRAY) and then recompile it？
For example, use the following command?
Catkin clean
Catkin build maplab

mfehr, 2019-03-07:
I'm actually running some color data of type BGR8 as we speak and all you need to do is this:
[image]
then
```bash
catkin build rovioli
```
Assuming your images are actually BGR8 and that's what's inside the image message encoding field.

gakkiHp, 2019-03-08:
Thank you very much, the problem has been solved, but I have another problem, I have modified maplab/common/vio-common/include/vio-common/internal/imu-measurements-buffer-inl.h file, then what is the command I need to recompile? Catkin build what?

gakkiHp, 2019-03-09:
thanks a lot, i have solved it according to what you say

closed 2019-03-09 by gakkiHp