ethz-asl/maplab #136: Performance on outdoors dataset

scharalambous3, 2019-03-07:
Dear maplab team.
Thank you for this great project!
I would like to consult your expertise to improve the performance of the framework on my dataset. Using the following calibrations, the sequential visualization of the VI map seems to diverge after a few seconds. Even though the path includes a curvature the output is a straight line with an extremely large distance (~10km). Also, as seen on the frame visualization screenshot the feature detection is very inaccurate. For your reference, I have also attached the rviz output, the terminal output, the map statistics and the print_camera_calibrations outputs. 
 
 
[image]
[image]
[image]
[image]

 
 
Upon changing doVECalibration to false, the distance decreased to 3km which is much better. Then, my next step would be to resolve the resolve the terminal warnings, and improve the feature tracking as well as the evaluated quality of the landmarks. Lastly, I have not modified the IMU sigmas yet (IMU used: Vectornav-200S)

For your reference I am also attaching the yaml files used and a sample rosbag 

https://drive.google.com/file/d/1MN68QPMDhyC0PXwKUENA6z421-wQnQoU/view?usp=sharing

Thanks in advance!
Stefanos Charalambous

antmusco, 2019-03-07:
Hi Stefanos,

What's your sensor setup look like? Rovio seems to be pretty sensitive to the timestamp of the IMU/Image data, depending on the update rate. I had a similar divergence problem using individual IMU/Camera sensors plugged into the same machine. If you are timestamping the images as they are recieved on the host (or the imu samples) then the latency/jitter is likely the cause for the divergence.

You can read more about it on the [Divergence](https://github.com/ethz-asl/rovio/wiki/Divergence) page of the Rovio wiki (note the last paragraph). #54 #56 also discuss the issue.

scharalambous3, 2019-03-08:
Thank you for your reply Anthony. My setup looks like this:
[image]
where I only use the middle camera and the IMU
The camera is generating the sync clock for IMU but dropping few frames
My concern is that neither the feature tracking nor the horizon level (it is unstable and exhibits a constant,slow CCW rotation) seem to be working correctly, indicating a potential problem in my calibration files which include the transformation matrices

For example, would more accurate  IMU sigmas resolve any of these issues, and if yes I will be pleased if you could recommend some examples to try

Thank you, 
Stefanos

antmusco, 2019-03-08:
If the rate of the IMU is approximately the rate of the camera (~30Hz), then tracking is likely to diverge quickly. AFAIK, Rovio (and all Visual-Inertial Odometry) relies on integrating the inertial data over a much smaller sample period (>100Hz) to get a good estimate for tracking when the next camera frame comes along. In my experience integrating inertial measurements at 20-50hz will not be sufficient to compute an accurate pose estimate for tracking.

Also, if you're seeing a constant, slow rotation then the gyro bias estimation (computed online) is likely not being estimated correctly. What is the behavior if you just leave the device on a static table and run Rovioli without any disturbance? Does the estimated pose slowly drift over time, or does it remain static?

Try the following, testing after each step to see if the situation improves.
1. Increase the IMU sample rate (if possible)
2. Re-run calibration using Kalibr.
    - Double check the camera calibration quality (<0.5 pixel re-projection error)
    - Double check that the IMU noise is sufficiently captured by the varianace estimate (just needs to be close).
    - Double check that the IMU/Camera extrinsic calibration makes sense (for debugging it just needs to be close - try displaying the frames in Matlab/ROS)
3. Manually decrease the gyro white noise sigma / increase the gyro random walk sigma (to try to capture the drift bias)

scharalambous3, 2019-03-09:
@antmusco Thanks again. Before I close this issue, can I contact you again through email or LI if I face any extreme difficulty in any of those steps?

closed 2019-03-12 by scharalambous3