ethz-asl/maplab #137: How does maplab do map merge?

ran5515, 2019-03-08:
Hi,

I have checked the maplab paper and some other research or discussing, only to know that to do merge, we need relocalization and scale match.

Is there any reference helpful to understand the algorithm used in maplab to merge map? Also, may anyone help to to locate where the merge code locate? (I am trying to find it)

Thanks!

antmusco, 2019-03-08:
The "merging" step (i.e. plugin "join_all_maps") doesn't really do anything but accumulate all missions from each map into a new map. This step doesn't perform any loop closure

As far as where it exists in the code, the plugin entry point is here:
https://github.com/ethz-asl/maplab/blob/618a0e0154d0c6e6cd84b5ac827140b5a9b84cb0/console-plugins/vi-map-basic-plugin/src/vi-map-basic-plugin.cc#L540
Which ends up calling this function which performs the merge:
https://github.com/ethz-asl/maplab/blob/618a0e0154d0c6e6cd84b5ac827140b5a9b84cb0/map-structure/vi-map/src/vi-map.cc#L44

The "Loop-Closure" plugin is where most of the magic happens, and that can be understood by looking into the `loopclosure/loop-closure-handler` algorithm directory, specifically the `loop-detector-node`

maplab/algorithms/loopclosure/loop-closure-handler/src/loop-detector-node.cc
      
For context, the loop closure plugin uses this detector to match all missions against one another in this plugin:
https://github.com/ethz-asl/maplab/blob/618a0e0154d0c6e6cd84b5ac827140b5a9b84cb0/console-plugins/loop-closure-plugin/src/vi-map-merger.cc#L31

ran5515, 2019-03-08:
@antmusco Wow, thanks for your detailed explain! I will check it carefully.

closed 2019-03-08 by ran5515