ethz-asl/maplab #138: Coordinate transformation on ROVIOLI positioning results and ground  truth of Euroc data

JWDu, 2019-03-08:
Hello everyone, When I tried to compare the ROVIOLI positioning results with the ground truth of the Euroc dataset, I found that the two did not correspond exactly. Is the coordinate system between the two different, and what is the conversion relationship between them?

mfehr, 2019-03-08:
Hi @JWDu 
Whenever you compare two trajectories you will inevitably need to align them first. Every estimator might define the origin of its trajectory differently (in camera coordinates, imu coordinates, might skip a couple of images etc.)
There are a lot of different methods and it depends on what you want to evaluate, you might want to do a global alignment with or without scale to evaluate global consistency or piece-wise alignment to analyze drift, etc.
Have a look at these evaluation tools, they should do both alignment and evaluation for you:
 * https://github.com/MichaelGrupp/evo
 * https://github.com/uzh-rpg/rpg_trajectory_evaluation

JWDu, 2019-03-09:
Thank you very much for your reply！ @mfehr 
I will try these tools.