ethz-asl/maplab #142: Localization Fails

onurgurisik, 2019-03-18:
Hello Everyone,

I am currently working on july_release of maplab. At the moment, my sensors frequencies as follow:

IMU: 100 Hz
Velocity : 100 Hz
CAM : 12 HZ 

I am using only one camera. It works fine at the beginning of localization, but after some time it gives me that error:

"Too few IMU measurements available between time 1552379995818016975[ns] and 1552379995818016976[ns]."

I checked my imu measurements and didn't notice any problem with them. By the way, there is no hardware synchronization between cam and imu. Do you think that this is the reason ?

I am looking forward your feedbacks.

Thank you

mfehr, 2019-04-12:
Hi @onurgurisik 
A good (hardware) time synchronization between IMU and camera is unfortunately a prerequisite to achieving a good performance with ROVIOLI/maplab. Your measurements might simply drift apart. If you are looking for an off-the-shelf sensor that you could use, we managed to get satisfactory results with the D435i and are currently testing the T265. Also I think we managed to get ok maps with the Mynteye S sensor as well after some tweaking.

alexcherpi, 2019-07-17:
Hi @mfehr,
Between the D435i, D430 OEM (only depth map), the T265 and the Strcture Core, which sensor do you recommend to use with ROVIOLI? Do you have any result from your tests?
Does ROVIOLI handle 3D poincloud input instead of 2D images? Do you plan to handle it?
Thanks

mfehr, 2019-07-23:
1. We have tested the D435i and T265, the former is a bit more challenging as the IMU data is not as nicely pre-calibrated, but we managed to get it to work, although not super stable. The latter works ok, except that the fisheye camera is a bit too extreme for ROVIO and it ends up rejecting more features than usual. So basically we aren't super happy with either, although the latter could be a nice option, but it will probably need some more work to adapt camera model and patch logic in ROVIO. No info in the structure core sensor.
2. We have tested one, but I have no quantitative evaluation of ROVIOLI vs the T265 tracking. The latter seems to work pretty well and we have used it for some applications, but the main issue there is that it's a black-box for us so we can't tinker with it if we have issues.
3. No and no, not for ROVIOLI, we are doing research to make maplab more multi-sensor capable, like lidars, but I can't say whether or when this will be open sourced.

closed 2019-07-23 by mfehr