ethz-asl/maplab #144: Map shrinking during optimization

onurgurisik, 2019-03-28:
Hello everyone,

I have some problem with map optimization. When I tried to optimize some of my maps with optvi or optv, map size dramatically shrinks and becomes meaningless.

Do you have any suggestion to prevent map shrinking?

Thank you ,
Best Regards,

mfehr, 2019-05-14:
Hi @onurgurisik,

One of the reasons we observed for this behavior is when the IMU data or sigmas is bad. When you calibrate the sensor with kalibr, have a look at the IMU error plots, which can provide some insight into whether your IMU data/sigmas are sound.

zoumaguanxin, 2020-04-15:
Hi @mfehr, 
  I have same problem. My used sensor is realsense d435i and  I use [imu_utils](https://github.com/gaowenliang/imu_utils) to calibrate the imu, `calibr` to cablirate the extrinsic parameters between camera and  imu. According to your suggestions, it is tested that bad sigmas cause terrible optimization result when using optvi, eg, the optimized pose trajectories is not smooth. But, in our experiment, optvi always shrink the map after about 10 iterations. For debug, I use optv to optimize my map, it shrinked the map also.  Can you give more advices to debug it? Is the imu baises provided by vio algorithm  important? Are there some FLAGS that make optimizing pose graph possible that only contain odometry constraints and lc constraints?

B-Meg, 2021-09-17:
@zoumaguanxin Could you find a way to fix the shrinking problem of the map optimization?

MrBurnst, 2022-02-08:
hi! are there any news on this? I'm having the same issue ...