ethz-asl/maplab #148: 6 DOF pose estimation using a visual-inertial sensor !

mechaliomar, 2019-05-09:
Hello, 
-I discovered your tool (maplab) recently. 
-I would like to know if there is ROVIO integrated in your maplab, because I want to use it for 6 DOF pose estimation using a visual-inertial (VI) sensor ??
-I want to acquire  images and  IMU data (accelerometer and gyroscope) from the VI sensor and then use ROVIO for the 6 DOF pose estimation. 
-Finally, the point cloud and the 6DOF pose is deleivred to VOXBLOX (a map builder). This map is used then for the path planning.

[image]
 

BEST REGARDS

mfehr, 2019-05-09:
Hi @mechaliomar 
Yes, the estimator we use to build maplab maps is called ROVIOLI and is an extension of ROVIO with localization and map building components. Please read the wiki and the maplab paper more information and instructions on how to use it. Once you have ROVIOLI running, use the instructions provided in voxblox and in your issue there (https://github.com/ethz-asl/voxblox/issues/258), which apply to ROVIOLI as well.

mechaliomar, 2019-05-09:
@mfehr  Thanks a lot for your replay. What do you think about OKVIS for 6 DOF pose estimation?

mfehr, 2019-05-10:
@mechaliomar That's quite a vague question :) I guess you should have a look at some review/survey papers if you want numbers. I can tell you is that at ASL most visual-inertial odometry onboard MAVs is done by ROVIO, which is mostly motivated by robustness as far as I know. It really depends on your application/data, e.g. a colleague of mine did an [evaluation](https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=8633393) on train datasets, and OKVIS did very well there.

closed 2019-05-10 by mfehr

mechaliomar, 2019-05-11:
@mfehr Thanks a lot.

wntun, 2020-08-04:
Hello, I have 6 DOF pose as odometry message from SLAM module (MSCKF-VIO). I don't know how to give that information to voxblox_node in the launch file. I only found 'transform' parameter and its type is TransformStamped. If it's the case, I would like to know how to convert the odometry to the transform stamped. 
Anyone please help me!