ethz-asl/maplab #151: Merging maps with the same ID

viktortuul, 2019-05-31:
Hi,

I have modified parts of ROVIOLI so that maps are being built sequentially during runtime, and developed an automatic map merger application (that uses maplab). This is for collaborative "sequential"-SLAM by having maps from multiple ROVIOLI applications being merged and optimized during runtime.

As of now, for each map merge process, the latest built VI-map from each active ROVIOLI application is loaded by the "automatic" maplab console with outputs a "global" map containing the individual maps from each ROVIOLI application. This map is during runtime loaded into each ROVIOLI application as well, so that the agents can localize on the global map.

However what I am trying to do now is to instead of only using the latest VI-maps in the merge process (pseudocode):
```
merged_map = maplab_commands(VI_map_ROVIOLI_1, VI_map_ROVIOLI2, ...)
```
to also include the latest merged map (to hopefully improve the global map quality/consistency between each merge), such as
```
merged_map = maplab_commands(merged_map_previous, VI_map_ROVIOLI_1, VI_map_ROVIOLI2, ...)
```

However, the problem is that the `merged_map_previous` already includes the same mission ID  of `VI_map_ROVIOLI_1` and `VI_map_ROVIOLI_2` (as the two VI-maps produced `merged_map_previous`) which renders this error:
```
NCamera with id 412eab8e4058621f7036b5e765dfe813 is already associated with mission 82ccaea0874ca1150d00000000000000.
```

Is there a known way of overcoming this problem, maybe assigning a new ID for each newly built map during run-time? Any advice how this could be done?

Thanks!

dymczykm, 2019-05-31:
hey @viktortuul ,
First of all, our IDs are randomly generated and have enough bits to guarantee they almost never clash. I think it's not an ID duplication here as well - it  rather seems like the ID of the NCamera (camera system) class that is used to store camera calibration is the same as you haved used the same camera calibration to build all the maps.

You can take a look here to see how the mission/map merging is done:
https://github.com/ethz-asl/maplab/blob/master/map-structure/vi-map/src/vi-map.cc#L68

mhkabir, 2019-06-10:
Hey @viktortuul, I'm working on a similar application as yours, and would love to speak with you about your experiences. I tried reaching out over email, but please let me know if there is a better way to reach you! You can reach me at mhkabir@mit.edu. Looking forward to the conversation!

GilSerrano, 2019-07-05:
Hi,

I have also been looking into this and was wondering how the vi-maps' mission IDs are set.  I am trying to merge several sequential vi-maps that have been created from the same dataset, so they have the same mission ID. 

From what I have done so far, it seems that it is not possible to merge 2 maps with the same mission ID, as I get the same error as in the first post:

`NCamera with id 412eab8e4058621f7036b5e765dfe812 is already associated with mission 08ec78f4ffffffff0f00000000000000.`

Despite the error stating that it is the Ncamera ID that is already associated with a certain mission, after looking more into the problem, it appears that this is not really an issue, since I am able to merge 2 maps with different mission IDs but with the same Ncamera ID.

My idea was to change the mission ID of the maps when they are being created, but I cannot figure out how to do this nor do I know if it would have any negative repercussions.

Any help is appreciated! Thanks :)

GilSerrano, 2019-07-22:
Thanks for the reply @dymczykm! Could you please give some more hints about how to remove a mission with a certain mission ID from the map?