ethz-asl/maplab #152: [ERROR]Realsense ZR300 rosnode timestamp error

maxwu8416, 2019-06-20:
hi ~
i'm try to use zr300 camera with nvidia tx2, when i try to launch the camera, the strange error display on terminal 

> E0620 22:05:46.581499  8614 zr300.cc:568] The timestamp of frame 1 of type FISHEYE did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
> E0620 22:05:46.614343  8614 zr300.cc:568] The timestamp of frame 2 of type FISHEYE did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
> E0620 22:05:46.647642  8614 zr300.cc:568] The timestamp of frame 3 of type FISHEYE did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
> E0620 22:05:46.681193  8614 zr300.cc:568] The timestamp of frame 4 of type FISHEYE did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
> E0620 22:05:46.714368  8614 zr300.cc:568] The timestamp of frame 5 of type FISHEYE did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
> E0620 22:05:46.747458  8614 zr300.cc:568] The timestamp of frame 6 of type FISHEYE did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
> E0620 22:05:46.780606  8614 zr300.cc:568] The timestamp of frame 7 of type FISHEYE did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
> E0620 22:05:46.823463  8614 zr300.cc:568] The timestamp of frame 1 of type INFRARED did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
> E0620 22:05:46.823626  8614 zr300.cc:568] The timestamp of frame 1 of type INFRARED2 did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
> E0620 22:05:46.844977  8614 zr300.cc:568] The timestamp of frame 1 of type DEPTH did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
> E0620 22:05:46.887357  8614 zr300.cc:568] The timestamp of frame 2 of type INFRARED did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
> E0620 22:05:46.887501  8614 zr300.cc:568] The timestamp of frame 2 of type INFRARED2 did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
> E0620 22:05:46.908155  8614 zr300.cc:568] The timestamp of frame 2 of type DEPTH did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
> E0620 22:05:46.935230  8614 zr300.cc:568] The timestamp of frame 2 of type COLOR did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!
> E0620 22:05:46.955919  8614 zr300.cc:568] The timestamp of frame 8 of type FISHEYE did not originate from the motion tracking microcontroller. The timestamp domain is set to: 0 (camera = 0, microcontroller = 1). Skipping frame!  
> 

i already read the code in zr300.cc:568, seems like somee error with timestamp calibr
when i run the librealsense example it works fine without error so i think it may not cause frome librealsense problem 

can anyone help me to figure it out ? 
thank you !!

mfehr, 2019-06-26:
Hi @maxwu8416
What the error means is that all this data (FISHEYE, DEPTH, COLOR, INFRARED, INFRARED2) are not timestamped using the more precise, synchronized motion tracking controller. In our version of the driver we therefore raise a warning because if the data is not properly synchronized things don't work (well). This synchronization should automatically happen once you set the zr300 to motion tracking mode, which we do in our ROS driver [here](https://github.com/ethz-asl/maplab_realsense/blob/master/src/zr300.cc#L50). But I remember that there were some issues with older devices/older firmware, which sometimes failed to do the synchronization properly and this error would be raised. You can try to disable this check/warning and see if it works or try to figure out if it's a problem of the version of the sensor.

maxwu8416, 2019-06-27:
hi  @mfehr 
thank you for the reply !!! refer your reply that more confirm my conclution.  
After few days suffer with this issue i think i have the pre-conclution, that i found out there has some imu data response  problem when using zr300 with tx2. That may cause there's no imu message output to the package (maplab_realsense) resulting the error message shows up.

closed 2019-06-27 by maxwu8416