ethz-asl/maplab #155: Build maplab Jetson Nano ARMv8 failure: opencv3_catkin

araujokth, 2019-07-26:
Hi,

I was trying to build maplab on the Jetson Nano ARMv8 processor. I first tried the master branch and it gets stuck building opencv3_catkin and it never continues from there, and there is no error. I then tried the `pre_release_public/march-2018` branch and the same thing happens. Any ideas what could be the problem? 

Thanks a lot for the help!

mfehr, 2019-07-26:
Hi @araujokth
Check if the compilation runs out of memory, try compiling using only one thread. There is an issue discussing compilation on a Jetson, maybe this helps.

araujokth, 2019-08-05:
Thanks! I will have a look at that. I had no problem with the out of memory though. I will update back when I do more tests!

araujokth, 2020-08-26:
Sorry for the very late update on this. We eventually managed to fix it and have this running on different NVidia Jetson devices (Nano, NX and AGX). The issue got solved by applying the following two additional patches to opencv3_catkin (march2018 branch)

  ```
PATCH_COMMAND patch -p0 < ${CMAKE_SOURCE_DIR}/fix_python_discovery.patch &&
		patch -p0 < ${CMAKE_SOURCE_DIR}/fix_lapack.patch &&
                patch -p0 < ${CMAKE_SOURCE_DIR}/fix_ccache_else_mismatch.patch
```

where fix_lapack.patch can be found [here](https://gitlab.seld.rnd.ericsson.se/txd_lmr/maplab/blob/develop/maplab_dependencies/3rdparty/opencv3_catkin/fix_lapack.patch) and the fix_ccache_else_mismatch.patch does the following

```diff
--- cmake/OpenCVCompilerOptions.cmake	2016-12-23 13:54:44.000000000 +0100
+++ cmake/OpenCVCompilerOptions.cmake.FIXED	2019-06-27 11:52:36.788324420 +0200
@@ -18,10 +18,10 @@
         message(STATUS "Unable to compile program with enabled ccache, reverting...")
         set_property(GLOBAL PROPERTY RULE_LAUNCH_COMPILE "${__OLD_RULE_LAUNCH_COMPILE}")
       endif()
+     endif()
     else()
       message(STATUS "Looking for ccache - not found")
     endif()
-  endif()
 endif()
 
 if((CMAKE_COMPILER_IS_CLANGCXX OR CMAKE_COMPILER_IS_CLANGCC OR CMAKE_COMPILER_IS_CCACHE) AND NOT CMAKE_GENERATOR MATCHES "Xcode")
```

closed 2020-08-26 by araujokth