ethz-asl/maplab #156: Error when install Maplab

trung1309vn, 2019-07-31:
Hi all. First, I want to thank you for this great project. I'm having some problems while installing the Maplab. My PC specs is as the following:
     - Operating System: Ubuntu 18.04 Bionic, 64 bit
     - ROS: melodic

Everything seem smoothly run until I use `catkin build maplab` . It sent me a bunch of errors, including brisk package error and a lot of abandonded packages. My catkin config and the error is as follow:
[image]

Errors     << brisk:make /home/aioz-trung-intern/maplab_ws/logs/brisk/build.make.010.log                 
/home/aioz-trung-intern/maplab_ws/src/maplab_dependencies/internal/ethzasl_brisk/brisk/src/brisk-feature-detector.cc: In function ‘void {anonymous}::RemoveInvalidKeyPoints(const cv::Mat&, std::vector<cv::KeyPoint>*)’:
/home/aioz-trung-intern/maplab_ws/src/maplab_dependencies/internal/ethzasl_brisk/brisk/src/brisk-feature-detector.cc:55:8: error: ‘function’ is not a member of ‘std’
   std::function<bool(const agast::KeyPoint& key_pt)> masking =
        ^~~~~~~~
/home/aioz-trung-intern/maplab_ws/src/maplab_dependencies/internal/ethzasl_brisk/brisk/src/brisk-feature-detector.cc:55:8: note: suggested alternative: ‘is_function’
   std::function<bool(const agast::KeyPoint& key_pt)> masking =
        ^~~~~~~~
        is_function
/home/aioz-trung-intern/maplab_ws/src/maplab_dependencies/internal/ethzasl_brisk/brisk/src/brisk-feature-detector.cc:55:17: error: expected primary-expression before ‘bool’
   std::function<bool(const agast::KeyPoint& key_pt)> masking =
                 ^~~~
/home/aioz-trung-intern/maplab_ws/src/maplab_dependencies/internal/ethzasl_brisk/brisk/src/brisk-feature-detector.cc:65:22: error: ‘masking’ was not declared in this scope
                      masking), keypoints->end());
                      ^~~~~~~
/home/aioz-trung-intern/maplab_ws/src/maplab_dependencies/internal/ethzasl_brisk/brisk/src/brisk-feature-detector.cc:65:22: note: suggested alternative: ‘mask’
                      masking), keypoints->end());
                      ^~~~~~~
                      mask
make[2]: *** [CMakeFiles/brisk.dir/src/brisk-feature-detector.cc.o] Error 1
make[1]: *** [CMakeFiles/brisk.dir/all] Error 2
make: *** [all] Error 2
cd /home/aioz-trung-intern/maplab_ws/build/brisk; catkin build --get-env brisk | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.........................................................................................................
Failed     << brisk:make                                      [ Exited with code 2 ]                     
Failed    <<< brisk                                           [ 0.7 seconds ]                            
Abandoned <<< aslam_cv_detector                               [ Unrelated job failed ]                   
Abandoned <<< aslam_cv_geometric_vision                       [ Unrelated job failed ]                   
Abandoned <<< aslam_cv_pipeline                               [ Unrelated job failed ]                   
Abandoned <<< aslam_cv_tracker                                [ Unrelated job failed ]                   
Abandoned <<< aslam_cv_visualization                          [ Unrelated job failed ]                   
Abandoned <<< aslam_serialization                             [ Unrelated job failed ]                   
Abandoned <<< console_common                                  [ Unrelated job failed ]                   
Abandoned <<< map_manager                                     [ Unrelated job failed ]                   
Abandoned <<< map_resources                                   [ Unrelated job failed ]                   
Abandoned <<< message_flow                                    [ Unrelated job failed ]                   
Abandoned <<< posegraph                                       [ Unrelated job failed ]                   
Abandoned <<< sensors                                         [ Unrelated job failed ]                   
Abandoned <<< statistics_plugin                               [ Unrelated job failed ]                   
Abandoned <<< vi_map                                          [ Unrelated job failed ]                   
Abandoned <<< dense_reconstruction_common                     [ Unrelated job failed ]                   
Abandoned <<< loopclosure_common                              [ Unrelated job failed ]                   
Abandoned <<< pose_graph_manipulation_plugin                  [ Unrelated job failed ]                   
Abandoned <<< simulation                                      [ Unrelated job failed ]                   
Abandoned <<< geometric_vision_algorithms                     [ Unrelated job failed ]                   
Abandoned <<< imu_integrator_rk4                              [ Unrelated job failed ]                   
Abandoned <<< ceres_error_terms                               [ Unrelated job failed ]                   
Abandoned <<< stereo_dense_reconstruction                     [ Unrelated job failed ]                   
Abandoned <<< vi_map_helpers                                  [ Unrelated job failed ]                   
Abandoned <<< map_optimization_legacy                         [ Unrelated job failed ]                   
Abandoned <<< vi_mapping_test_app                             [ Unrelated job failed ]                   
Abandoned <<< landmark_triangulation                          [ Unrelated job failed ]                   
Abandoned <<< vio_common                                      [ Unrelated job failed ]                   
Abandoned <<< visualization                                   [ Unrelated job failed ]                   
Abandoned <<< feature_tracking                                [ Unrelated job failed ]                   
Abandoned <<< landmark_manipulation_plugin                    [ Unrelated job failed ]                   
Abandoned <<< map_optimization_legacy_plugin                  [ Unrelated job failed ]                   
Abandoned <<< map_sparsification                              [ Unrelated job failed ]                   
Abandoned <<< maplab_console                                  [ Unrelated job failed ]                   
Abandoned <<< online_map_builders                             [ Unrelated job failed ]                   
Abandoned <<< vi_map_basic_plugin                             [ Unrelated job failed ]                   
Abandoned <<< vi_map_visualization_plugin                     [ Unrelated job failed ]                   
Abandoned <<< vocabulary_tree                                 [ Unrelated job failed ]                   
[... 9 lines cut ...]
Abandoned <<< loop_closure_plugin                             [ Unrelated job failed ]                   
Abandoned <<< map_anchoring_plugin                            [ Unrelated job failed ]                   
Abandoned <<< map_optimization                                [ Unrelated job failed ]                   
Abandoned <<< map_optimization_plugin                         [ Unrelated job failed ]                   
Abandoned <<< map_sparsification_plugin                       [ Unrelated job failed ]                   
Abandoned <<< mapping_workflows_plugin                        [ Unrelated job failed ]                   
Abandoned <<< rovioli                                         [ Unrelated job failed ]                   
Abandoned <<< vi_map_data_import_export_plugin                [ Unrelated job failed ]                   
Abandoned <<< vi_map_summarization_plugin                     [ Unrelated job failed ]                   
Abandoned <<< voxblox_interface                               [ Unrelated job failed ]                   
Abandoned <<< pmvs_interface                                  [ Unrelated job failed ]                   
Abandoned <<< dense_reconstruction_plugin                     [ Unrelated job failed ]                   
Finished  <<< aslam_cv_matcher                                [ 0.1 seconds ]                            
Finished  <<< aslam_cv_triangulation                          [ 0.1 seconds ]                            
_________________________________________________________________________________________________________
Errors     << maplab_common:make /home/aioz-trung-intern/maplab_ws/logs/maplab_common/build.make.010.log 
/usr/bin/ld: cannot find -lreadline
collect2: error: ld returned 1 exit status
make[2]: *** [/home/aioz-trung-intern/maplab_ws/devel/lib/libmaplab_common.so] Error 1
make[1]: *** [CMakeFiles/maplab_common.dir/all] Error 2
make: *** [all] Error 2
cd /home/aioz-trung-intern/maplab_ws/build/maplab_common; catkin build --get-env maplab_common | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.........................................................................................................
Failed     << maplab_common:make                              [ Exited with code 2 ]                     
Failed    <<< maplab_common                                   [ 1.0 seconds ]

mfehr, 2019-07-31:
Hi @trung1309vn

Compatibility with 18.04 on the master branch (there are some newer release branches that should fix it) is not quite there, but also not too difficult to do, there should be only a few, repeating compilation issues throughout the code such as:
 * all these "function" related errors: just add #import <functional> to those files
 
The last package is missing a library called readline, should have been installed IMO during the installation instructions but maybe check and install it 

Hope that helps

trung1309vn, 2019-07-31:
Hi @mfehr , thank you for your answer. I'll try your suggestion asap.

keithito, 2019-08-02:
It looks like there's an open PR to fix the missing `#include <functional>` in ethzasl_brisk:
https://github.com/ethz-asl/ethzasl_brisk/pull/109

trung1309vn, 2019-08-03:
Hello @keithito , I really appeciate for your answer. I'll check the link as you suggested.

trung1309vn, 2019-08-05:
I have successfully built Maplab by following [](https://github.com/ethz-asl/ethzasl_brisk/pull/109) as suggested.
Thank you all for helping me.

closed 2019-08-05 by trung1309vn