ethz-asl/maplab #160: Publishing raw images and their pose

araujokth, 2019-08-14:
Hi,

My goal was to be able to publish raw images and their respective pose computed by maplab. 

I have successfully been able to publish keyframes and their respective pose by modifying [VIO_UPDATES in data-publisher-flow.cc](https://github.com/ethz-asl/maplab/blob/88d8587958c6902e7304a5ef7d88e867b30924b3/applications/rovioli/src/data-publisher-flow.cc#L92) and reading and publishing the output of vio_update->keyframe_and_imudata->nframe->getFrame() and I also modified the rviz-visualization-sink so that I publish each image with the vio_update->timestamp_ns as the timestamp. However, I would also like to be able to publish the frame and its respective pose for each captured frame. 

I noticed that the VIO_UPDATE flow is only running whenever a keyframe is available, while ROVIO_ESTIMATES publishes for every frame, so my thinking was to change it to be able to get the frames here. But was wondering if you had any suggestions on how to do this in the best way so that I make sure that I publish a frame with its respective pose.

Thanks again for the help!

mfehr, 2019-08-14:
Check out this flag here, it throttles the VisualNFrames that are assembled, I think if you increase this to a value higher than the image frequency it should yield one VioUpdate per image.

``--vio_nframe_sync_max_output_frequency_hz``

araujokth, 2019-08-14:
oh, that it much easier and much cleaner than I thought :) had completely missed that flag! I will give that a try and thanks a lot again for the help!

araujokth, 2019-08-14:
It worked! Thanks a lot again!

closed 2019-08-14 by araujokth