ethz-asl/maplab #165: mapconsole use freak false！

wangyuanbiubiubiu, 2019-10-06:
When I'm using freak with rovio, I use lc on the map panel and I report an error. Then I delete all my dat files to see if it's because the mapconsole has a brisk configuration file loaded. And I found out that it was. I would like to ask, how can I make my mapconsole also use freak descriptor? This is my configuration file.The picture shows I lack a brisk configuration file, but I actually meant freak

[image]
[image]

mfehr, 2019-10-06:
Hi @wangyuanbiubiubiu 
Have you tried telling the console which descriptor you want to use with the `--feature_descriptor_type` flag? You can either do that at startup (the setting is persistent for all commands executed later) or when launching the lc command (but then it's not persistent).

wangyuanbiubiubiu, 2019-10-06:
> Hi @wangyuanbiubiubiu
> Have you tried telling the console which descriptor you want to use with the `--feature_descriptor_type` flag? You can either do that at startup (the setting is persistent for all commands executed later) or when launching the lc command (but then it's not persistent).

I've fixed this problem by adding this command while the terminal is running. But doesn't mapconsole have its own script file? Similar to the rovio script file, so I can add this configuration directly to the file

mfehr, 2019-10-06:
Glad it worked.

closed 2019-10-06 by mfehr