ethz-asl/maplab #168: Strip Voxblox from Maplab to expedite ROS2 port

anamud, 2019-10-16:
Is Voxblox a show-stopping dependency for Maplab? 

I am porting Maplab to ROS2 and would like to avoid porting Voxblox. I am mostly interested in VIO with LC on sparse maps. I read  [this](https://github.com/ethz-asl/voxblox/issues/260) humorous description of maplab vs voxblox and am assuming Voxblox is not required for my case. Is this assumption correct?

mfehr, 2019-10-16:
Hi @anamud 
Short answer: it's doable, but it's as easy as just dropping the dense reconstruction plugin.

Algorithmically maplab doesn't need voxblox to perform any of it's VIO tasks and it is only used in the dense reconstruction plugin. The issue is the resource system, which supports storing and attaching voxblox maps to the VIMaps, this part is quite deeply embedded. We also use some of the voxblox types for other dense reconstruction algorithms. So you would need to:
 - drop the dense reconstruction plugin, or part of it
 - check the resource system and remove all voxblox resources. This is gonna be a bit all over the place, but shouldn't be hard.

Keep us posted about the ROS2 upgrade, we'd be interested to know how it goes.

anamud, 2019-10-16:
Hi @mfehr, thanks for the helpful info. I will proceed with stripping voxblox out now.

closed 2019-10-16 by anamud

nroussel00, 2024-12-23:
Hi @anamud. I know this issue is 5 years old and it's a very long shot, but I'm myself interested on running maplab with ROS2...
Any chance you published a public fork somewhere? :) I'm in the same case as you described, working ROS2-based VIO and mainly want to add LC on top