ethz-asl/maplab #173: Can maplab output comlete state estimation?

pedropgusmao, 2019-11-04:
Hello, 

Besides the actual pose, is it possible for Maplab to provide an estimate of angular velocity, linear velocity, linear acceleration and IMU biases?
 
Thanks,

Pedro

dymczykm, 2019-11-19:
@pedropgusmao we are reading the state of ROVIO and publishing it here:
https://github.com/ethz-asl/maplab/blob/master/applications/rovioli/src/rovio-flow.cc#L188
You could tap on these values to read the linear velocity and IMU biases.

For the angular velocity/linear velocity you can look here:
https://github.com/ethz-asl/maplab/blob/master/applications/rovioli/src/feature-tracking.cc#L18
Where we subscribe to IMU data and biases to debias them and use for feature tracking.

Hope that helps.

pedropgusmao, 2019-11-19:
Hi @dymczykm , thanks for your answer.
I was wondering also if it is possible also output those values after loop-closer and optimization. Maybe from the console?

Thanks,

araujokth, 2020-08-26:
Yes, you can do that via the console using the command `export_trajectory_to_csv` and this is what is exported https://github.com/ethz-asl/maplab/blob/88d8587958c6902e7304a5ef7d88e867b30924b3/console-plugins/vi-map-data-import-export-plugin/src/export-vertex-data.cc#L65

closed 2020-09-07 by pedropgusmao