ethz-asl/maplab #175: Use maptracking

wangyuanbiubiubiu, 2019-11-14:
I found out that maplab didn't do a maptracking relocation, I did one. Break it down into three steps. Step 1: use uniform motion model to project the map points to the current frame, and then conduct description sub-matching. PNP solution. Step 2: if unsuccessful, select adjacent frames for word bag search and conduct a small range of relocation. Step 3: if not, return to Global mode for PNP. Is that your thinking?

One more question, does the current maplab code support reconstruction with an external depth map?

bohell, 2020-11-18:
I am quite interested in your work. How's the result? Is it reliable?

wangyuanbiubiubiu, 2020-11-18:
> I am quite interested in your work. How's the result? Is it reliable?

Yes, it is reliable. This method has significantly increased recall rates

closed 2020-11-18 by wangyuanbiubiubiu