ethz-asl/maplab #176: How can i use Candidate March 2018

wangyuanbiubiubiu, 2019-11-15:
I wanted to use the code under this branch, but I ran into some problems compiling it. I would like to ask if I need any other operations after checkout?

mfehr, 2019-11-15:
Hi @wangyuanbiubiubiu 
Please have a look at the PR description  https://github.com/ethz-asl/maplab/pull/55

closed 2019-11-15 by mfehr

wangyuanbiubiubiu, 2019-11-16:
I did follow your steps, but the following error occurred.
[image]

I thought I would download the package after using your command, but it didn't
[image]

wangyuanbiubiubiu, 2019-11-16:
> Hi @wangyuanbiubiubiu
> Please have a look at the PR description #55

My question is put in the above reply. How can I solve this problem？Thx

wangyuanbiubiubiu, 2019-11-16:
[image]

wangyuanbiubiubiu, 2019-11-16:
This is my procedure, can you help me to see what went wrong
step1.git clone https://github.com/ethz-asl/maplab.git --recursive
step2.git checkout pre_release_public/march-2018
step3.git clone https://github.com/ethz-asl/maplab_dependencies --recursive
step4.git checkout pre_release_public/march-2018
step5.git submodule init
step6.git submodule update --recursive
step7.catkin build maplab