ethz-asl/maplab #178: How to use resource-importer？

wangyuanbiubiubiu, 2019-11-21:
I had attached the depth to my vi-map by following the steps on the wiki, but when I ran "create_tsdf_from_depth_resource" I still couldn't recreate it successfully。
[image]
[image]

wangyuanbiubiubiu, 2019-11-21:
And I found that if I
Step1: rosrun maplab_console maplab_console
Step2: load --map_folder xx
Step3: SDR balabala
Step4: create_tsdf_from_depth_resource
That's fine.
But when I quit, I can't go back and try to use the depth of the previous binocular calculation, for example
Step1: rosrun maplab_console maplab_console
Step2: load --map_folder xx
Step3: create_tsdf_from_depth_resource
That's not going to work

wangyuanbiubiubiu, 2019-11-21:
[image]

mfehr, 2019-11-21:
Hi @wangyuanbiubiubiu 
Did you save the map after running stereo? If you don't then the mp doesn't know it is supposed to have esources when you load it again

wangyuanbiubiubiu, 2019-11-21:
> Hi @wangyuanbiubiubiu
> Did you save the map after running stereo? If you don't then the mp doesn't know it is supposed to have esources when you load it again

Yes, when I saved it, I did re-execute create_tsdf_from_depth_resource

However, the same problem will still occur when I use my own depth using resource-importer, that is, the reconstruction will not work. Do I need to give the depth map exactly the same timestamp as each node in my map?

wangyuanbiubiubiu, 2019-11-21:
This is the resources data generated by resource-importer. I noticed that dense_depth_resource_input_type will affect the program. In my case, how much should this parameter be set?
[image]

wangyuanbiubiubiu, 2019-11-21:
I set it to 8, and it works

closed 2019-11-21 by wangyuanbiubiubiu

wangyuanbiubiubiu, 2019-11-23:
yeah，I forget save it，the issue now fix，thx！