ethz-asl/maplab #180: run rovioli --Segmentatiion fault

zhizunhu, 2019-11-30:
Hi!
I have finished catkin build maplab, but when i try to run rovioli ,terminal shows Segmentation fault. I try to debug it in VSCode , it shows:
[image]

mfehr, 2019-12-16:
Please add more information, as requested in the issue template.

closed 2019-12-16 by mfehr

JingeTu, 2020-04-24:
I encountered the same bug. It is a segmentation fault during program initialization. May be something wrong with gflags.
When I run `LD_DEBUG=files ~/maplab_ws/devel/lib/rovioli/rovioli`, it shows all *.so files loaded successfully, but segmentation fault when calling init of libglog.so.0.
When I run `valgrind --leak-check=full --show-leak-kinds=all -v ~/maplab_ws/devel/lib/rovioli/rovioli`, valgrind shows as follows:
```
==24717== Process terminating with default action of signal 11 (SIGSEGV)
==24717==  Bad permissions for mapped region at address 0x68DD318
==24717==    at 0xA3C0B31: __static_initialization_and_destruction_0 (logging.cc:134)
==24717==    by 0xA3C0B31: _GLOBAL__sub_I_logging.cc (logging.cc:2103)
==24717==    by 0x40106C9: call_init.part.0 (dl-init.c:72)
==24717==    by 0x40107DA: call_init (dl-init.c:30)
==24717==    by 0x40107DA: _dl_init (dl-init.c:120)
==24717==    by 0x4000C69: ??? (in /lib/x86_64-linux-gnu/ld-2.23.so)
```
It shows there is something wrong when initializing global static variables.

I have tried rebuilding whole maplab_ws, setting gflags_catkin and glog_catkin linking to local installed gflags and glog, using newest gflags and glog. All not working.

But the same time built maplab_console is OK, I can `run maplab_console maplab_console` and do the procedures as maplab paper given.