ethz-asl/maplab #194: Implement Maplab with my video

EmreOzkose, 2019-12-16:
Implement Maplab with my video
---

Hi! Basically, I am trying to implement Maplab with my video. Here is my pipeline:

1. Convert video to bagfile (done)
2. Create map with rovioli (occur an error)
3. Load the map with Maplab

During following this pipeline, I achieved (1), but I could not achieve (2). 

1. I created bagfile with this script from [there](https://stackoverflow.com/questions/31432870/how-do-i-convert-a-video-or-a-sequence-of-images-to-a-bag-file).
2. When I run `rosrun rovioli tutorial_euroc save_folder /home/MY_NAME/PycharmProjects/empyt/bagfile.bag`, output is: 

> I1216 14:42:36.252480 22628 rovio-factory.cc:169] Loading ROVIO configuration template: /home/eramas/maplab_ws/src/maplab/applications/rovioli/share//rovio_default_config.info
Set distortion parameters (Radtan) to: k1(-0.283408), k2(0.0739591), k3(0), p1(0.000193595), p2(1.76187e-05)
Registered filter state update callback.
I1216 14:42:36.493990 22628 map-builder-flow.cc:31] Set VIMap folder to: save_folder_8
I1216 14:42:36.494524 22628 rovioli-node.cc:105] Starting data source...
I1216 14:42:36.494570 22658 datasource-rosbag.cc:194] Rosbag playback finished!
I1216 14:42:36.494544 22628 rovioli-app.cc:136] 
Message delivery queues:
subscriber-node               queue-topic                   queue-id                      num elements                  
MapBuilderFlow                LOCALIZATION_RESULT           8d0d..0000                    0                             
MapBuilderFlow                TRACKED_NFRAMES_AND_IMU       6300..0000                    0                             
DataPublisherFlow             VIO_UPDATES                   5a33..0000                    0                             
DataPublisherFlow             RAW_VIMAP                     3507..0000                    0                             
DataPublisherFlow             ROVIO_ESTIMATES               d927..0000                    0                             
DataPublisherFlow             LOCALIZATION_RESULT           f01e..0000                    0                             
SyncedNFrameThrottlerFlow     TRACKED_NFRAMES_AND_IMU       32c0..0000                    0                             
FeatureTrackingFlow           ROVIO_ESTIMATES               5e87..0000                    0                             
RovioFlow                     IMAGE_MEASUREMENTS            dbfb..0000                    0                             
FeatureTrackingFlow           SYNCED_NFRAMES_AND_IMU        a968..0000                    0                             
RovioFlow                     IMU_MEASUREMENTS              e1c0..0000                    0                             
MapBuilderFlow                ROVIO_ESTIMATES               8807..0000                    0                             
MapBuilderFlow                VIO_UPDATES                   3aee..0000                    0                             
ImuCameraSynchronizerFlow     IMU_MEASUREMENTS              294e..0000                    0                             
RovioFlow                     LOCALIZATION_RESULT           810d..0000                    0                             
ImuCameraSynchronizerFlow     IMAGE_MEASUREMENTS            5945..0000                    0                             
I1216 14:42:37.494823 22628 rovioli-node.cc:110] Closing data source...
W1216 14:42:37.495124 22628 map-builder-flow.cc:101] Map is empty; nothing will be saved.
I1216 14:42:37.495610 22628 rovioli-node.cc:110] Closing data source...

Map is empty. I coul not understand why? How can I debug this issue?
Thank you in advance.

mfehr, 2019-12-16:
Hi @EmreOzkose 
Do you have IMU data?

EmreOzkose, 2019-12-16:
No, I just gave a video which has mp4 format to the script for creating bagfile and then this bagfile became input to `rosrun rovioli tutorial_euroc`

mfehr, 2019-12-18:
Probably a time sync issue, please have a look at the FAQ on some answers related to our sensor requirements

https://github.com/ethz-asl/maplab/wiki/FAQ#q-what-sensors-are-recommended-for-maplab

EmreOzkose, 2019-12-19:
I solved these problems by making long videos (probably realted with feature extraction) and writing first IMU data to bagfile (writing first /imu0/ instead of /cam0/image_raw). However, when I have a video that contains a loop, map is still creted like a line. That means,  map could not be created well.

I guess like you. There is a problem with creation of bagfile. Unfortunately, I do not have a tango-enabled device. Hence my videos are made by a smartphone. My pipeline is basically: 
1. Make video with a smartphone (with IMU data)
2. Convert this video to bagfile with that script (as I mention before, this script is mostly from [here](https://stackoverflow.com/questions/31432870/how-do-i-convert-a-video-or-a-sequence-of-images-to-a-bag-file) ):

```
import time
import sys
import cv2
from utils import read_data, read_bag_file
import math

import ros
from ros import rosbag
import roslib, rospy
from cv_bridge import CvBridge
import csv

roslib.load_manifest("geometry_msgs")
from geometry_msgs.msg import Quaternion, Vector3
roslib.load_manifest("sensor_msgs")
from sensor_msgs.msg import Image
from sensor_msgs.msg import Imu

def read_data(csv_path, video_path):

    csv_file = open(csv_path)
    csv_reader = csv.reader(csv_file, delimiter=",")
    header_dict = {}
    row_list = []

    for line_count, row in enumerate(csv_reader):
        row_dict = {}
        if line_count == 0:
            for i, r in enumerate(row):
                header_dict[r] = i
        else:
            row_dict["Acc X"] = float(row[header_dict["Acc X"]])
            row_dict["Acc Y"] = float(row[header_dict["Acc Y"]])
            row_dict["Acc Z"] = float(row[header_dict["Acc Z"]])

            row_dict["gyro_x"] = float(row[header_dict["gyro_x"]])
            row_dict["gyro_y"] = float(row[header_dict["gyro_y"]])
            row_dict["gyro_z"] = float(row[header_dict["gyro_z"]])
            row_list.append(row_dict)

    cap = cv2.VideoCapture(video_path)
    cap.set(cv2.CAP_PROP_FPS, 15)
    ret = True

    frame_id = 0
    frame_and_imu = []

    frame_list = []
    while True:
        ret, frame = cap.read()
        if ret is False:
            break

        frame = cv2.resize(frame, (752, 480))
        frame_list.append(frame)

    cap.release()

    print(len(frame_list))
    print(len(row_list))

    frame_list_len = len(frame_list)
    row_list_len = len(row_list)

    a = np.array(range(frame_list_len), dtype=np.float) / frame_list_len
    b = np.array([int(i) for i in np.array(range(row_list_len), dtype=np.float) / row_list_len])

    a = cv2.resize(a, (1, b.shape[0]))
    a = [int(i) for i in a*frame_list_len]

    for i, j in zip(a, b):
        frame_imu_dict = dict()
        try:
            frame_imu_dict["frame"] = frame_list[i]
            frame_imu_dict["imu_data"] = row_list[j]

            frame_and_imu.append(frame_imu_dict)
        except IndexError:
            pass

    return frame_and_imu

def video2bag(video_path, bag_name):
    bag = rosbag.Bag(bag_name, 'w')
    cb = CvBridge()

    video_name = "20191219_122640"
    frame_and_imu_list = read_data(csv_path="video_plus_imu/{}.csv".format(video_name),
                                   video_path="video_plus_imu/{}.mp4".format(video_name))

    for frame_id, frame_and_imu in enumerate(frame_and_imu_list):
        print(frame_id)
        frame = frame_and_imu["frame"]
        imu_data = frame_and_imu["imu_data"]

        stamp = rospy.Time.from_sec(time.time())

        image = cb.cv2_to_imgmsg(frame, encoding='bgr8')

        image.header.seq = frame_id
        image.header.stamp = stamp
        image.header.frame_id = "imu4"

        # We currently do not know the covariance of the sensors with each other
        cov = [0, 0, 0, 0, 0, 0, 0, 0, 0]
        quat1 = [0.0, 0.0, 0.0, 1.0]
        quat = Quaternion(quat1[0] / 16384.0, quat1[1] / 16384.0, quat1[2] / 16384.0, quat1[3] / 16384.0)

        # # Normalize the quaternion and accelerometer values
        quatNorm = math.sqrt(quat.x * quat.x + quat.y * quat.y + quat.z * quat.z + quat.w * quat.w)
        # normQuat = Quaternion(quat.x / quatNorm, quat.y / quatNorm, quat.z / quatNorm, quat.w / quatNorm)
        normQuat = Quaternion(0.0, 0.0, 0.0, 1.0)

        acc = [imu_data["Acc X"], imu_data["Acc Y"], imu_data["Acc Z"]]
        gyro = [imu_data["gyro_x"], imu_data["gyro_y"], imu_data["gyro_z"]]

        normAcc = Vector3(acc[0] / 2048.0, acc[1] / 2048.0, acc[2] / 2048.0)
        normGyro = Vector3(gyro[0] / 16.0, gyro[1] / 16.0, gyro[2] / 16.0)
        imu = Imu(image.header, normQuat, cov, normGyro, cov, normAcc, cov)

        bag.write("/imu0", imu, stamp)
        bag.write("/cam0/image_raw", image, stamp)

    bag.close()
```

So, is it a proper way, or we must have a tango-enabled device? 
There are some application where you can take video with IMU data. Is there a way with a smartphone setup with these apps?

araujokth, 2020-08-13:
Unfortunately maplab wont work if you dont have a good hardware platform with good time synch, which is hard to obtain from a smartphone. I would very much recommend that you try to get your hands on the Intel D435i or D455 cameras which are pretty cheap to run maplab instead of using a smartphone