ethz-asl/maplab #224: Status of "Release Candidate March 2018"

araujokth, 2020-03-04:
I was wondering what was the status of the "Release Candidate March 2018" PR https://github.com/ethz-asl/maplab/pull/55, as if you are planning to release it in the near future, or this is just a dead branch which is no longer in your plans to be released? I was trying to understand why the PR had not been successful as well as it says tests and checks failed but there is no details on what has failed.

Regarding the further testing that you mention in the PR, is there something in particular you think should be tested which you could use some help with?

I was also wondering about the comment made on the TODO list "Investigate and fix slightly deteriorated performance of map alignment and merging", as I could not find any details about where these problems were detected? 

Thanks for the help!

mfehr, 2020-03-04:
Hi @araujokth 
That's a good question. There are two major things that prevented us at that time from merging this candidate. The first is that we would need to check and update all tutorials, update some of the data we offer alongside them. It changes some things a bit, as documented in the PR description. So if we did that we better be sure this version is worth the effort. The second issue is that we noticed that one of the tutorial wouldn't work as well anymore (the multi-session, staircase), the alignment wouldn't work properly anymore with the same commands and we couldn't find an explanation that made sense within the time we had available. Hence we were hesitant to merge it.
Since until recently we were developing in a private copy we simply didn't have the resources to stop, transfer the changes, address these things, test that we still get the numbers in the paper, update the doc and wiki, properly assign version numbers, etc. And since people were actively using master, which performed well and they didn't seem to mind it wasn't a priority. And then... well time flies.
We switched to the public version now, at least for the core (develop branch), so this will get easier and might happen, or not.
I know this isn't a very satisfying answer, but that's how it is :)

araujokth, 2020-03-06:
Thanks for the quick reply @mfehr ! I totally understand the dilemma :) So basically no plans yet if and when you guys will make another release right?

Just let me know if I can be of any help for testing, etc.!

closed 2020-03-06 by araujokth