ethz-asl/maplab #247: Error when installing Maplab, need help!

bennywijaya55, 2020-04-13:
Profile:                              default
Extending:          [cached] /opt/ros/kinetic
Workspace:                       /home/benny/Desktop/Maplab
---------------------------------------------------------------
Build Space:        [exists]   /home/benny/Desktop/Maplab/build
Devel Space:        [exists]  /home/benny/Desktop/Maplab/devel
Install Space:      [unused] /home/benny/Desktop/Maplab/install
Log Space:          [exists]   /home/benny/Desktop/Maplab/logs
Source Space:       [exists] /home/benny/Desktop/Maplab/src
DESTDIR:            [unused] None
---------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
---------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------------
Workspace configuration appears valid.
---------------------------------------------------------------
[build] Found '105' packages in 0.0 seconds.                                                                                                                                                               
[build] Package table is up to date.                                                                                                                                                                       
Starting  >>> catkin_simple                                                                                                                                                                                
Starting  >>> doxygen_catkin                                                                                                                                                                               
Starting  >>> kindr                                                                                                                                                                                        
Starting  >>> libnabo                                                                                                                                                                                      
Starting  >>> suitesparse                                                                                                                                                                                  
Finished  <<< doxygen_catkin                                  [ 0.1 seconds ]                                                                                                                              
Finished  <<< suitesparse                                     [ 0.1 seconds ]                                                                                                                              
Finished  <<< catkin_simple                                   [ 0.1 seconds ]                                                                                                                              
Starting  >>> eigen_catkin                                                                                                                                                                                 
Starting  >>> gflags_catkin                                                                                                                                                                                
Starting  >>> hand_eye_calibration                                                                                                                                                                         
Starting  >>> lpsolve_catkin                                                                                                                                                                               
Starting  >>> maplab_test_data                                                                                                                                                                             
Starting  >>> metis_catkin                                                                                                                                                                                 
Starting  >>> opencv3_catkin                                                                                                                                                                               
Starting  >>> protobuf_catkin                                                                                                                                                                              
Starting  >>> yaml_cpp_catkin                                                                                                                                                                              
Finished  <<< kindr                                           [ 0.1 seconds ]                                                                                                                              
Finished  <<< eigen_catkin                                    [ 0.1 seconds ]                                                                                                                              
Starting  >>> opengv                                                                                                                                                                                       
Finished  <<< protobuf_catkin                                 [ 0.1 seconds ]                                                                                                                              
Finished  <<< gflags_catkin                                   [ 0.1 seconds ]                                                                                                                              
Starting  >>> glog_catkin                                                                                                                                                                                  
Finished  <<< maplab_test_data                                [ 0.1 seconds ]                                                                                                                              
Finished  <<< opencv3_catkin                                  [ 0.1 seconds ]                                                                                                                              
Starting  >>> cv_bridge                                                                                                                                                                                    
Finished  <<< hand_eye_calibration                            [ 0.1 seconds ]                                                                                                                              
Starting  >>> end_to_end_common                                                                                                                                                                            
Finished  <<< lpsolve_catkin                                  [ 0.1 seconds ]                                                                                                                              
Finished  <<< yaml_cpp_catkin                                 [ 0.4 seconds ]                                                                                                                              
Finished  <<< metis_catkin                                    [ 0.1 seconds ]                                                                                                                              
Finished  <<< libnabo                                         [ 0.3 seconds ]                                                                                                                              
Finished  <<< glog_catkin                                     [ 0.1 seconds ]                                                                                                                              
[... cut ...]    
[build]   Failed:    1 packages failed.                                                                                                                                                                    
[build] Runtime: 8.8 seconds total.                                                                                                                                                                        
[build] Note: Workspace packages have changed, please re-source setup files to use them.

bennywijaya55, 2020-04-13:
First, I want to say thank you for this amazing project. I'm having some problems while installing the Maplab. My PC specs is as the following:
- Operating System: Ubuntu 16.04, 64 bit
- ROS: kinetic

bennywijaya55, 2020-04-13:
This is the log file of build.make.000.log

[  3%] [34m[1mRunning C++ protocol buffer compiler on proto/maplab-common/id.proto[0m
[  6%] [34m[1mRunning C++ protocol buffer compiler on proto/maplab-common/eigen.proto[0m
[35m[1mScanning dependencies of target maplab_common[0m
[  9%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/binary-serialization.cc.o[0m
[ 12%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/breakpoints.cc.o[0m
[ 16%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/combinatorial.cc.o[0m
[ 19%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/condition.cc.o[0m
[ 22%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/cubic-spline.cc.o[0m
[ 25%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/delayed-notification.cc.o[0m
[ 29%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/feature-descriptor-ref.cc.o[0m
[ 32%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/file-lock.cc.o[0m
[ 35%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/file-logger.cc.o[0m
[ 38%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/file-system-tools.cc.o[0m
[ 41%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/geometry.cc.o[0m
[ 45%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/global-coordinate-tools.cc.o[0m
[ 48%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/gnuplot-interface.cc.o[0m
[ 51%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/gravity-provider.cc.o[0m
[ 54%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/histograms.cc.o[0m
[ 58%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/multi-threaded-progress-bar.cc.o[0m
[ 61%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/progress-bar.cc.o[0m
[ 64%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/proto-serialization-helper.cc.o[0m
[ 67%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/python-interface.cc.o[0m
[ 70%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/shared-gflags.cc.o[0m
[ 74%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/sigint-breaker.cc.o[0m
[ 77%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/stringprintf.cc.o[0m
[ 80%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/test/testing-entrypoint.cc.o[0m
[01m[K/home/benny/Desktop/Maplab/src/maplab/common/maplab-common/src/stringprintf.cc:[m[K In function ‘[01m[Kvoid common::StringAppendV(std::__cxx11::string*, const char*, __va_list_tag*)[m[K’:
[01m[K/home/benny/Desktop/Maplab/src/maplab/common/maplab-common/src/stringprintf.cc:55:14:[m[K [01;35m[Kwarning: [m[Kcomparison between signed and unsigned integer expressions [-Wsign-compare]
   if (result < sizeof(space)) {
[01;32m[K              ^[m[K
[ 83%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/threading-helpers.cc.o[0m
[ 87%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/tridiagonal-matrix.cc.o[0m
[ 90%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/src/unique-id.cc.o[0m
[ 90%] [34m[1mRunning C++ protocol buffer compiler on proto/maplab-common/eigen.proto[0m
[ 90%] [34m[1mRunning C++ protocol buffer compiler on proto/maplab-common/id.proto[0m
[ 93%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/compiled_proto/maplab-common/eigen.pb.cc.o[0m
[ 96%] [32mBuilding CXX object CMakeFiles/maplab_common.dir/compiled_proto/maplab-common/id.pb.cc.o[0m
[100%] [32m[1mLinking CXX shared library /home/benny/Desktop/Maplab/devel/lib/libmaplab_common.so[0m
/usr/bin/ld: cannot find -lreadline
collect2: error: ld returned 1 exit status
CMakeFiles/maplab_common.dir/build.make:899: recipe for target '/home/benny/Desktop/Maplab/devel/lib/libmaplab_common.so' failed
make[2]: *** [/home/benny/Desktop/Maplab/devel/lib/libmaplab_common.so] Error 1
CMakeFiles/Makefile2:2775: recipe for target 'CMakeFiles/maplab_common.dir/all' failed
make[1]: *** [CMakeFiles/maplab_common.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

mfehr, 2020-04-13:
Hi @bennywijaya55 
The readline library is missing. Did you follow the instructions to install the system dependecies? https://github.com/ethz-asl/maplab/wiki/Installation-Ubuntu#install-required-system-packages

bennywijaya55, 2020-04-13:
thanks! bug solved! 
was missing this step before.. 

sudo apt install autotools-dev ccache doxygen dh-autoreconf git liblapack-dev libblas-dev libgtest-dev libreadline-dev libssh2-1-dev pylint clang-format-3.8 python-autopep8 python-catkin-tools python-pip python-git python-setuptools python-termcolor python-wstool libatlas3-base --yes

sudo pip install requests

closed 2020-04-13 by bennywijaya55