ethz-asl/maplab #278: Error in installing Maplab (libtbb.so)

dadung, 2020-09-17:
Hello, thank you for publishing this framework. 

I am trying to install maplab, but it has a error, I guess the problem is the confusion between libraries in my machine & anaconda, below is the error, can you please help me resolve it? Thank you very much in advance!

______________________________________________________________________________________________________
Errors     << brisk:make /home/dzungdoan/DzungDoan/libs/maplab_ws/logs/brisk/build.make.002.log                                                                                                            
//home/dzungdoan/anaconda3/lib/libtbb.so.2: undefined reference to `__cxa_init_primary_exception@CXXABI_1.3.11'
//home/dzungdoan/anaconda3/lib/libtbb.so.2: undefined reference to `std::__exception_ptr::exception_ptr::exception_ptr(void*)@CXXABI_1.3.11'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/dzungdoan/DzungDoan/libs/maplab_ws/devel/lib/brisk/camera] Error 1
make[1]: *** [CMakeFiles/camera.dir/all] Error 2
make: *** [all] Error 2

______________________________________________________________________________________________________
Errors     << rovio:make /home/dzungdoan/DzungDoan/libs/maplab_ws/logs/rovio/build.make.002.log                                                                                                            
//home/dzungdoan/anaconda3/lib/libtbb.so.2: undefined reference to `__cxa_init_primary_exception@CXXABI_1.3.11'
//home/dzungdoan/anaconda3/lib/libtbb.so.2: undefined reference to `std::__exception_ptr::exception_ptr::exception_ptr(void*)@CXXABI_1.3.11'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/dzungdoan/DzungDoan/libs/maplab_ws/devel/lib/rovio/feature_tracker_node] Error 1
make[1]: *** [CMakeFiles/feature_tracker_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
//home/dzungdoan/anaconda3/lib/libtbb.so.2: undefined reference to `__cxa_init_primary_exception@CXXABI_1.3.11'
//home/dzungdoan/anaconda3/lib/libtbb.so.2: undefined reference to `std::__exception_ptr::exception_ptr::exception_ptr(void*)@CXXABI_1.3.11'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/dzungdoan/DzungDoan/libs/maplab_ws/devel/lib/rovio/rovio_rosbag_loader] Error 1
make[1]: *** [CMakeFiles/rovio_rosbag_loader.dir/all] Error 2
//home/dzungdoan/anaconda3/lib/libtbb.so.2: undefined reference to `__cxa_init_primary_exception@CXXABI_1.3.11'
//home/dzungdoan/anaconda3/lib/libtbb.so.2: undefined reference to `std::__exception_ptr::exception_ptr::exception_ptr(void*)@CXXABI_1.3.11'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/dzungdoan/DzungDoan/libs/maplab_ws/devel/lib/rovio/rovio_node] Error 1
make[1]: *** [CMakeFiles/rovio_node.dir/all] Error 2
make: *** [all] Error 2

araujokth, 2020-09-17:
I guess you are sourcing and setting up anaconda in your bashrc? Remove it, open a new terminal build maplab, and then just source things for anaconda when you need it.

dadung, 2020-09-20:
Hello, thank you for your suggestion, I've solved the problem, but there is still another new problem. Below is what I've done

I edited bashrc using the command
`gedit ~/.bashrc`

Then, I commented lines related to Anaconda
[image]

Then, the problem above is solved (Yay! :smile: ). But it has a new problem (also the confusion about anaconda3) :cry: 
```
Errors     << maplab_console:make /home/dzungdoan/DzungDoan/libs/maplab_ws/logs/maplab_console/build.make.002.log                                                                                          
/home/dzungdoan/anaconda3/lib/libharfbuzz.so.0: undefined reference to `FT_Get_Var_Blend_Coordinates'
/home/dzungdoan/anaconda3/lib/libpng16.so.16: undefined reference to `inflateValidate@ZLIB_1.2.9'
/home/dzungdoan/anaconda3/lib/libfontconfig.so.1: undefined reference to `FT_Done_MM_Var'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/dzungdoan/DzungDoan/libs/maplab_ws/devel/lib/maplab_console/batch_runner] Error 1
make[1]: *** [CMakeFiles/batch_runner.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/dzungdoan/anaconda3/lib/libharfbuzz.so.0: undefined reference to `FT_Get_Var_Blend_Coordinates'
/home/dzungdoan/anaconda3/lib/libpng16.so.16: undefined reference to `inflateValidate@ZLIB_1.2.9'
/home/dzungdoan/anaconda3/lib/libfontconfig.so.1: undefined reference to `FT_Done_MM_Var'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/dzungdoan/DzungDoan/libs/maplab_ws/devel/lib/maplab_console/maplab_console] Error 1
make[1]: *** [CMakeFiles/maplab_console.dir/all] Error 2
make: *** [all] Error 2

```
Here is my `PATH` and `LD_LIBRARY_PATH` (displayed by `printenv`)
`PATH=/opt/ros/kinetic/bin:/home/dzungdoan/bin:/home/dzungdoan/.local/bin:/home/dzungdoan/anaconda3/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin`

`LD_LIBRARY_PATH=/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu`

Could you please provide me a suggestion for this problem? Thank you very much! :pray:

dadung, 2020-09-20:
I've solved the problem, just need to restart the installation procedure from the beginning

closed 2020-09-20 by dadung